hippo5329/linorobot2_hardware #2: Zio I2C driver

samuk, 2024-02-11:
Thanks for your work on this! Much appreciated.

When building for Pico I get this build error
` pio run -e pico_zio -t upload
Processing pico_zio (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: rpipico; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico.html
PLATFORM: Raspberry Pi RP2040 (1.9.0+sha.ac45b10) > Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, pico-debug, picoprobe, raspberrypi-swd)
PACKAGES: 
 - framework-arduinopico @ 1.30301.0+sha.3cc5ac1 
 - tool-mklittlefs-rp2040-earlephilhower @ 5.100300.230216 (10.3.0) 
 - tool-openocd-rp2040-earlephilhower @ 5.100300.230216 (10.3.0) 
 - tool-rp2040tools @ 1.0.2 
 - toolchain-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Installing pyyaml with pip at PlatformIO environment
/home/sam/.platformio/penv/bin/python -m pip install pyyaml
Requirement already satisfied: pyyaml in /home/sam/.platformio/penv/lib/python3.11/site-packages (6.0.1)
Installing markupsafe==2.0.1 with pip at PlatformIO environment
/home/sam/.platformio/penv/bin/python -m pip install markupsafe==2.0.1
Requirement already satisfied: markupsafe==2.0.1 in /home/sam/.platformio/penv/lib/python3.11/site-packages (2.0.1)
Configuring rpipico with transport serial
Downloading micro-ROS dev dependencies
KeyError: '':
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 609:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/sam/.platformio/platforms/raspberrypi@src-ff76a3915224135aafad379817f41edd/builder/main.py", line 253:
    target_elf = env.BuildProgram()
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 62:
    env.ProcessProjectDeps()
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piobuild.py", line 152:
    plb = env.ConfigureProjectLibBuilder()
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 1183:
    project.install_dependencies()
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 968:
    if _is_builtin(spec):
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 960:
    for lb in self.env.GetLibBuilders():
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 1096:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 60:
    obj = globals()[clsname](env, path, verbose=verbose)
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 153:
    self.process_extra_options()
  File "/home/sam/.platformio/penv/lib/python3.11/site-packages/platformio/builder/tools/piolib.py", line 301:
    self.env.SConscript(
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 609:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/sam/.platformio/packages/tool-scons/scons-local-4.6.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/sam/Desktop/hardware/linorobot2_hardware-master/firmware/.pio/libdeps/pico_zio/micro_ros_platformio/extra_script.py", line 157:
    build_microros()
  File "/home/sam/Desktop/hardware/linorobot2_hardware-master/firmware/.pio/libdeps/pico_zio/micro_ros_platformio/extra_script.py", line 99:
    builder.run('{}/metas/{}'.format(main_path, selected_board_meta), cmake_toolchain.path, microros_user_meta)
  File "/home/sam/Desktop/hardware/linorobot2_hardware-master/firmware/.pio/libdeps/pico_zio/micro_ros_platformio/microros_utils/library_builder.py", line 63:
    self.download_dev_environment()
  File "/home/sam/Desktop/hardware/linorobot2_hardware-master/firmware/.pio/libdeps/pico_zio/micro_ros_platformio/microros_utils/library_builder.py", line 96:
    for repo in Sources.dev_environments[self.distro]:
=========================================================================================== [FAILED] Took 2.86 seconds ===========================================================================================

Environment    Status    Duration
-------------  --------  ------------
pico_zio       FAILED    00:00:02.857
`

I'm actually keen to run the Zio driver with a ESP32-S3 board, so I'll try and work out how to run the Zio code under a ESP32 instead

hippo5329, 2024-02-11:
Try, sudo apt install dfu-util
I will add a configuration for esp32_zio later.

hippo5329, 2024-02-11:
I will add more details in the wiki.

samuk, 2024-02-11:
Thanks. Even after installing dfu-util and with board_microros_distro = humble I still see the same error when building Pico.

When building for esp32_zio using this [esp32_zio.config.h](https://raw.githubusercontent.com/rosmo-robot/linorobot2_hardware/master/config/custom/esp32_zio_config.h) file and this [Plaftormio.ini](https://github.com/rosmo-robot/linorobot2_hardware/blob/f58688e92d3abae4ee2baeea71103c700d460f1e/firmware/platformio.ini#L114) I can successfully build for serial transport. 

I haven't yet successfully built with wifi transport but I'll keep playing with it.

hippo5329, 2024-02-11:
Please uncomment this to enable wifi
// #define WIFI_AP_LIST {{"WIFI_SSID", "WIFI_PASSWORD"}, {NULL}}

hippo5329, 2024-02-11:
I would suggest based your esp32_zio_config on esp32_config.
Add change the zio motor pins.

    #define PWM_BITS 12                         // PWM Resolution of the microcontroller
    #define PWM_FREQUENCY 100                   // PWM Frequency
    #define PCA_BASE 100
    #define MOTOR1_PWM  100
    ...
    #define BOARD_INIT_LATE { \
    setLevel(MOTOR_STBY_1, HIGH); \
    setLevel(MOTOR_STBY_2, HIGH); \
    }

t

hippo5329, 2024-02-11:
First copy esp32_config to esp32_zio_config.
Follow the wiki. Build, upload and connect to wifi agent.
Run the ros2 topic echo.
Then add the zio stuff.

hippo5329, 2024-02-11:
The earlier building error on micro-ros,

https://github.com/hippo5329/linorobot2_hardware/wiki#10-platformio-stop-building-at-downloading-micro-ros-dev-dependencies

closed 2024-02-16 by hippo5329