hippo5329/linorobot2_hardware #14: Adding MPU6050 to the firmware

AdibSuid, 2024-05-26:
Sorry, I wanna ask why is the micro ros agent stuck when I uncommented #define USE_MPU6050_IMU? But working fine when I commented it

hippo5329, 2024-05-26:
Please open an Q&A in the discussion. https://github.com/hippo5329/linorobot2_hardware/discussions
"Issue" should be used for bug reporting.

It seems the IMU is not initialized correctly in your case. Check the serial message with "pio device monitor -e xxx" . It will run "i2cdetect" at the beginning of setup(). If MPU6050 is detected, it should be like this.

    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

Otherwise, the MPU6050 is not detected. Check your I2C pin assignment and wiring.

hippo5329, 2024-05-26:
## I2C bus pull up requirement

The esp32/pico require external 2K Ohm resisters pull up on the I2C SDA SCL line to the 3.3V output of the esp32/pico. Because I2C interfaces of esp32/pico can be assign to other I/O pins, there are no I2C pull up on the esp32/pico modules.

The teensy40 needs these pull up, too. 

CY521 has on board 3.3V LDO. Connect the VCC of CY521/MPU6050 to 5V.

hippo5329, 2024-05-26:
I push an update. "teensy: need Wire.begin for i2cdetect". Please pull.

```
# save your changes
git add .
git commit -m "my config"
# pull the update
git pull
```

AdibSuid, 2024-05-27:
> Please open an Q&A in the discussion. https://github.com/hippo5329/linorobot2_hardware/discussions
> "Issue" should be used for bug reporting.
> 
> It seems the IMU is not initialized correctly in your case. Check the serial message with "pio device monitor -e xxx" . It will run "i2cdetect" in the beginning of setup(). If MPU6050 is detected, it should be like this.
> 
> ```
>     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- -- 
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
> 70: -- -- -- -- -- -- -- --
> ```
> 
> Otherwise, the MPU6050 is not detected. Check your I2C pin assignment and wiring.
> 
> 

I see, alright thankss🙌🏻😁😇

hippo5329, 2024-05-28:
I added more details on micro-ros agent connection test and bringup launch. Please check the wiki.

closed 2024-06-19 by hippo5329