hippo5329/linorobot2_hardware #15: Waiting for robot_description to be published on robot_description topic

Leesanhang, 2024-05-27:
Hi there, I am currently doing my project on ROS2. Here are some of the hardware I am using:
Mecanum base, esp32-WROOM-32 microcontroller, Cytron MDD10A (Generic 1) motor drivers, A2M8 RPLidar, and a raspberry Pi 4B as the main computer. 

I have followed the steps in this page exactly up until the "test_acc" part. One small issue that I did encounter was "test_sensors" part. The values are not changing even when I tilt my robot and so on. (But i guess its fine since the Gyro isn't that important for my project).

The main issue that I've been facing for the past 2 weeks was this: 
when I run:
**ros2 launch linorobot2_bringup bringup.launch.py**
or
**ros2 launch linorobot2_bringup bringup.launch.py base_serial_port:=/dev/ttyUSB0 lidar_serial_port:=/dev/ttyUSB1 micro_ros_baudrate:=921600**

it gives me this in the terminal:
[image]

it just waits infinitely for the robot_description to be published on the robot_description topic.

If it helps, this is my urdf file for my robot:

`<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:property name="base_length" value="0.49" />
  <xacro:property name="base_width" value="0.39" />
  <xacro:property name="base_height" value="0.06" />
  
  <xacro:property name="wheel_radius" value="0.0723" />
  <xacro:property name="wheel_width" value="0.05" />

  <xacro:property name="laser_pose">
    <origin xyz="-0.0475 0 0.125" rpy="0 0 0"/>
  </xacro:property>
  
  <material name="grey">
        <color rgba="0.5 0.5 0.5 1" />
    </material>

    <material name="blue">
        <color rgba="0 0 0.5 1" />
    </material>
    
    <material name="pink">
        <color rgba="1 0.75 0.8 1" />
    </material>
    
    

    <link name="base_footprint"/>
    
    <joint name="base_joint" type="fixed">
        <parent link="base_footprint"/>
        <child link="base_link"/>
        <origin xyz="0 0 0.2" rpy="0 0 0"/>
    </joint>
    
    

    <link name="base_link">
        <visual>
            <geometry>
                <box size="${base_length} ${base_width} ${base_height}" />
            </geometry>
            <origin xyz="0 0 0" rpy="0 0 0" />
            <material name="blue" />
        </visual>
    </link>
    
    <joint name="base_scan_joint" type="fixed">
        <parent link="base_link"/>
        <child link="base_scan"/>
        <origin xyz="0 0 0.06" rpy="0 0 0"/>
    </joint>
    
    <link name="base_scan">
        <visual>
            <geometry>
                <cylinder radius="0.0375" length="0.04"/>
            </geometry>
            <origin xyz="0 0 0.04" rpy="0 0 0" />
            <material name="pink" />
        </visual>
    </link>
    
    
    
    <joint name="base_left_back_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="left_back_wheel"/>
        <origin xyz="-0.22 0.225 -0.01" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
    </joint>

    <link name="left_back_wheel">
        <visual>
            <geometry>
                <cylinder radius="${wheel_radius}" length="${wheel_width}"/>
            </geometry>
            <origin xyz="0 0 0" rpy="1.57 0 0" />
            <material name="grey" />
        </visual>
    </link>
    
    <joint name="base_right_back_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="right_back_wheel"/>
        <origin xyz="-0.22 -0.225 -0.01" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
    </joint>

    <link name="right_back_wheel">
        <visual>
            <geometry>
                <cylinder radius="0.05" length="0.0723"/>
            </geometry>
            <origin xyz="0 0 0" rpy="1.57 0 0" />
            <material name="grey" />
        </visual>
    </link>
    
    <joint name="base_left_front_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="left_front_wheel"/>
        <origin xyz="0.22 0.225 -0.01" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
    </joint>

    <link name="left_front_wheel">
        <visual>
            <geometry>
                <cylinder radius="0.05" length="0.0723"/>
            </geometry>
            <origin xyz="0 0 0" rpy="1.57 0 0" />
            <material name="grey" />
        </visual>
    </link>
    
    <joint name="base_right_front_wheel_joint" type="continuous">
        <parent link="base_link"/>
        <child link="right_front_wheel"/>
        <origin xyz="0.22 -0.225 -0.01" rpy="0 0 0"/>
        <axis xyz="0 1 0"/>
    </joint>

    <link name="right_front_wheel">
        <visual>
            <geometry>
                <cylinder radius="0.05" length="0.0723"/>
            </geometry>
            <origin xyz="0 0 0" rpy="1.57 0 0" />
            <material name="grey" />
        </visual>
    </link>
    
    
</robot>
`
Any help to solve this issue is much appreciated, Thank youu!

hippo5329, 2024-05-27:
1. Both esp32 and lidar use the same device names as /dev/ttyUSBx, 0 or 1. It is very easy to mixed up. Disconnect the lidar first. Only the esp32 is connected.
2. Please run the micro-ROS agent connection test. https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-serial-transport
3. imu is not enabled by default in esp32_config.h. Which imu do you use? You will need to uncomment the line, eg, #define USE_MPU6050_IMU. You should see the i2cdetect. https://github.com/hippo5329/linorobot2_hardware/wiki#i2cdetect-debug-message . And the imu reading in test_sensors should change as you move. You should also see the /imu/data topic changing in step 2.
4. gyro is important. Ekf filter will fuse the wheel odometry and yaw from imu.
5. You should install the linorobot2 with "a2", which is for rplidar a2. eg. bash install_linorobot2.bash 2wd a2
6. You will need the by-path names to specify which serial port is esp32 and lidar. https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-ld19-lidar
7. In the bring up messages, you should see many messages from "micro ros agent", "joint state publisher", "robot state publisher" and rplidar. In your photo, the micro-ros agent is not connected yet and the lidar did not start.

hippo5329, 2024-05-28:
https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-lidar-on-the-robot-computer

hippo5329, 2024-05-28:
I added more details on micro-ros agent connection test and bringup launch. Please check the wiki.

Leesanhang, 2024-05-29:
Hi again, I am quite new to ROS so there are quite a few things that I am still unfamiliar with. For the past day, I have been trying to do step 2 and this is the output I am getting.

[image]

The open device errors are due to me unplugging the esp and plugging it back in.

I didn't face this problem before I started the changes at [https://github.com/hippo5329/linorobot2_hardware/wiki#custom-configuration-file---recommended](url)

I am quite unsure on how to proceed now.

hippo5329, 2024-05-29:
No worries. I am new to ROS2 and I keep learning, too.
Please follow my suggested work flow. In another directory, clone my fork again. Follow this,
https://github.com/hippo5329/linorobot2_hardware/wiki#install-firmware-source
Do not change the source. Build, upload and test connection.
https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-serial-transport

hippo5329, 2024-05-29:
The access mode of ttyUSB0 might be wrong. You might miss the step,

Install [PlatformIO udev rules](https://docs.platformio.org/en/latest/core/installation/udev-rules.html).

curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules
sudo service udev restart
sudo usermod -a -G dialout $USER
sudo usermod -a -G plugdev $USER

Then unplug and plug esp32.

$ ls /dev/ttyU* -l
crw-rw-rw- 1 root dialout 188, 0 May 29 15:22 /dev/ttyUSB0

Leesanhang, 2024-05-30:
So, I've tried doing everything from scratch again and I've passed the connection test here: [https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-serial-transport](url)

And now, I am currently doing step 3 in your earlier comment: [https://github.com/hippo5329/linorobot2_hardware/issues/15#issuecomment-2133987557](url)

This is the output I am getting:
[image]
(I've tried changing the baudrate because I've encountered similar problems in earlier projects related to Arduino.)

Is there any way I can complete this project without implementing the IMU?

hippo5329, 2024-05-30:
Good. Please tell me which IMU you use. And show me the configuration of your IMU. The micro-ROS middle ware is running on the serial. So you will see random chars because the communication is binary, not ascii code. Keep the baud rate at 921600. You will need to press reset button on the esp32 to see the messages right after boot.

hippo5329, 2024-05-30:
You should run the test_sensors. There is no micro-ros protocol in test_sensors. Press reset button on the esp32 to see the boot message.

Leesanhang, 2024-05-30:
I am using an MPU6050. My IMU is connected to an Arduino 5V and GND for the time being, and the SDA and SCL are connected to pins 21 and 22 of my ESP respectively.

hippo5329, 2024-05-30:
Why Arduino? Connect 5V to 5V on esp32 module.

hippo5329, 2024-05-30:
Connect 5V of MPU6050 to 5V on esp32.  Connect gnd of MPU6050 to gnd on esp32.

Leesanhang, 2024-05-30:
Sorry but I dont actually have a 5V pin on my ESP. I am using this dev board: [https://i0.wp.com/circuits4you.com/wp-content/uploads/2018/12/ESP32-Pinout.jpg?resize=604%2C373&ssl=1](url)

hippo5329, 2024-05-30:
Please check with a multimeter the Vin of esp32. If you see 5-4.5V, then use the Vin. Otherwise use 3.3V output of esp32.

hippo5329, 2024-05-30:
There is a LDO 3.3V on the GY521. The MPU6050 chip uses 3.3V. It is fine.

Leesanhang, 2024-05-30:
I've connected all the relevant pins as you instructed in previous comments. 5V and GND of IMU to Vin and GND of ESP.
Running "test_sensors" gives me the same output - random characters.

hippo5329, 2024-05-30:
pio device monitor -e esp32 -b 921600

Leesanhang, 2024-05-30:
I've tried defining the baudrate for earlier comments too. Same, using "-b 921600" but the outputs are the same. Same case for this "test_sensors" command.

Leesanhang, 2024-05-30:
I'm really not sure why is it outputting these characters. I had no such issues earlier, even when running "test_motors", "test_sensors", and "test_acc".

hippo5329, 2024-05-30:
cd test_motors
pio run -e esp32 -t upload
pio device monitor -e esp32
Test with a bare esp32 like yours. I will add a mpu6050 later.

ACC  0.00  0.00  0.00 GYR  0.00  0.00  0.00 MAG  0.00  0.00  0.00
 BAT  0.45V RANGE  0.00m
��␀�␀���␀�␀␀␀�␀�����␀��␀␀␀␀␀␀␀␀␀␀␀␀␀�␀␀␀�␀␀␀␀␀␀␀␀�␀␀␀␀␀␀␀␀�␀���␀��␀��␀␀�␀��␀␀�␀␀␀��␀�␀��␀�␀␀␀␀��␀␀␀␀␀�␀␀���␀�������␀�␀␀���␀�␀␀����␀␀����␀��␀����␀�␀����␀�␀␀�␀␀␀�������␀��␀�␀␀␀␀��␀␀�␀␀␀␀␀�␀�␀��␀���␀␀����␀␀��␀��␀��␀������␀␀�␀␀��␀��␀��␀␀����␀␀��␀��␀�␀����␀�␀��␀␀�␀�␀����␀��␀␀��␀���␀��␀��␀��␀�␀�␀␀␀␀␀␀�␀␀␀�␀��␀���␀␀�����␀��␀�␀␀␀␀���␀���␀���␀␀�␀�␀�␀���������␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀�␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀␀␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀�␀��␀�␀␀���␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀␀��␀␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀���␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀��␀␀�␀�������␀␀���␀�␀�␀␀␀��␀␀��␀�␀�␀␀␀�␀�␀������␀�����␀␀␀␀��␀��␀��␀�␀�␀␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀��␀␀␀␀�␀␀�␀␀�␀␀␀␀�␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀�␀␀␀�␀␀�␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀�␀�␀␀␀␀␀��␀␀␀␀␀␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀�␀␀�␀␀␀�␀␀�␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀�␀�␀␀␀␀␀␀␀␀␀␀�␀␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀␀�␀␀␀␀␀��␀␀�␀��␀�␀��␀��␀␀��␀�␀␀␀␀␀�␀�␀␀␀␀␀␀␀␀␀���␀␀␀�␀␀��␀␀�␀     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

ACC  0.00  0.00  0.00 GYR  0.00  0.00  0.00 MAG  0.00  0.00  0.00
 BAT  1.15V RANGE  0.00m
ACC  0.00  0.00  0.00 GYR  0.00  0.00  0.00 MAG  0.00  0.00  0.00
 BAT  0.66V RANGE  0.00m
ACC  0.00  0.00  0.00 GYR  0.00  0.00  0.00 MAG  0.00  0.00  0.00
 BAT  0.58V RANGE  0.00m
ACC  0.00  0.00  0.00 GYR  0.00  0.00  0.00 MAG  0.00  0.00  0.00
 BAT  0.63V RANGE  0.00m

hippo5329, 2024-05-30:
Remove previous libs build.
rm -rf ~/.platformio
rm -rf linorobot2_hardware/firmware/.pio
rm -rf linorobot2_hardware/test_motors/.pio
rm -rf linorobot2_hardware/test_sensors/.pio

Then build again.

Leesanhang, 2024-05-30:
Sorry but what does this error mean?
bash: /home/leesanhang/.platformio/penv/bin/pio: No such file or directory

hippo5329, 2024-05-30:
It might be link. It is not important. Rebuild and test again.

hippo5329, 2024-05-30:
Oh, I deleted pio. Need to reinstall platformio.

curl -fsSL -o get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
python3 get-platformio.py

Leesanhang, 2024-05-30:
Yup, I just realised platformio was deleted. Got it, I reinstalled and I am trying to build now.

hippo5329, 2024-05-30:
with mpu9250, and press reset button.

ACC  0.06 -0.04 11.17 GYR -0.07 -0.07 -0.07 MAG  0.00  0.00  0.00
 BAT  4.35V RANGE  0.00m
��␀�␀���␀�␀␀␀�␀�����␀��␀␀␀␀␀␀␀␀␀␀␀␀␀�␀␀␀�␀␀␀␀␀␀␀␀�␀␀␀␀␀␀␀␀�␀���␀��␀��␀␀�␀��␀␀�␀␀␀��␀�␀��␀�␀␀␀␀��␀␀␀␀␀�␀␀���␀�������␀�␀␀���␀�␀␀����␀␀����␀��␀����␀�␀����␀�␀␀�␀␀␀�������␀��␀�␀␀␀␀��␀␀�␀␀␀␀␀�␀�␀��␀���␀␀����␀␀��␀��␀��␀������␀␀�␀␀��␀��␀��␀␀����␀␀��␀��␀�␀����␀�␀��␀␀�␀�␀����␀��␀␀��␀���␀��␀��␀��␀�␀�␀␀␀␀␀␀�␀␀␀�␀��␀���␀␀�����␀��␀�␀␀␀␀���␀���␀���␀␀�␀�␀�␀���������␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀�␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀␀␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀�␀��␀�␀␀���␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀␀��␀␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀␀�␀���␀����␀␀�␀␀�␀��␀�␀␀␀␀�␀␀␀␀��␀␀�␀�������␀␀���␀�␀�␀␀␀��␀␀��␀�␀�␀␀␀�␀�␀������␀�����␀␀␀␀��␀��␀��␀�␀�␀␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀��␀␀␀␀�␀␀�␀␀�␀␀␀␀�␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀�␀␀␀�␀␀�␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀�␀�␀␀␀␀␀��␀␀␀␀␀␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀�␀␀�␀␀␀�␀␀�␀��␀␀�␀␀������␀�␀␀�␀�␀␀␀␀�␀�␀␀␀␀␀␀␀␀␀␀�␀␀␀␀␀␀�␀␀����␀�␀��␀�␀�␀␀␀�␀␀␀␀␀��␀␀�␀��␀�␀��␀��␀␀��␀�␀␀␀␀␀�␀�␀␀␀␀␀␀␀␀␀���␀␀␀�␀␀��␀␀�␀     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

ACC  0.60 -1.51 10.80 GYR  0.07  0.03  0.07 MAG  0.00  0.00  0.00
 BAT  5.17V RANGE  0.00m
ACC  1.01 -0.85 10.66 GYR -0.11 -0.09 -0.11 MAG  0.00  0.00  0.00
 BAT  0.68V RANGE  0.00m

Leesanhang, 2024-05-30:
I'm still getting this
[image]
when I run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
`

hippo5329, 2024-05-30:
Do you use ubuntu alone or are you using ubuntu inside windows?
How about test_sensors? Press reset button to see the the message.

hippo5329, 2024-05-30:
check, ls /dev/ttyU* -l

hippo5329, 2024-05-30:
Try another usb cable.

Leesanhang, 2024-05-30:
I actually just tried and restart it a few times and now it works (weirdly). Now, Im trying to do the rest, up until test_acc, will keep you updated.

hippo5329, 2024-05-30:
upload, device monitor, micro-ros agent can not run at the same time. The use the same serial port. Only one at a time.

Leesanhang, 2024-05-30:
So I tried building and testing some of the commands.

Running 
`cd firmware
pio run -e myrobot`
for [https://github.com/hippo5329/linorobot2_hardware/wiki#i2cdetect-debug-message](url)
Gives me:
[image]

and

Running
`cd test_motors
pio run -e myrobot -t upload
pio device monitor -e myrobot`
for [https://github.com/hippo5329/linorobot2_hardware/wiki#test-the-motors-and-encoders](url)
Gives me:
[image]

One more thing I'm curios is that, it takes me very long to build, roughly 15 mins where it will be "Building micro-ROS library".

Leesanhang, 2024-05-30:
> Do you use ubuntu alone or are you using ubuntu inside windows? How about test_sensors? Press reset button to see the the message.

I use ubuntu 22.04 in my Raspberry Pi 4B 4GB.

Leesanhang, 2024-05-30:
One more update before logging off for a few hours:
I went ahead and checked some of the src folders, in the firmware.cpp file, for example: the firmware.cpp folder in test_motors. I realised that the baudrates are set to 115200. I changed it to 921600 and this is the output that I am getting:
[image]
Exact same outputs when I run the test_motors commands and the IMU checking commands. The 68 appears when I press the reset button on my ESP.

Leesanhang, 2024-05-30:
I figured I should show you this too:
[image]
The values aren't changing even when I'm moving the gyro around. Oh yes, and I tried changing the baudrate to 921600 for test_sensors as well.

hippo5329, 2024-05-30:
please uncomment the line #define USE_MPU6050_IMU to enable mpu6050

hippo5329, 2024-05-30:
env [esp32] use esp32_config.h, pio run -e esp32 -t upload
env [myrobot] use myrobot_config.h, pio run -e myrobot -t upload

don't mix up

hippo5329, 2024-05-31:
I updated the workflow in wiki to avoid the trouble to create custom configuration file for beginners.

hippo5329, 2024-05-31:
Please post photos of your esp32 and wiring.

Leesanhang, 2024-05-31:
Hi again, I've been looking around in the linorobot2_hardware folder.
I narrowed the micro-ROS connection issue down to this:
In the /config/custom/esp32config.h file and the /config/lino_base_config.h file, I replaced the following lines to this:
`//define your robot' specs here
#define MOTOR_MAX_RPM 222                   // motor's max RPM
#define MAX_RPM_RATIO 1.0                  // max RPM allowed for each MAX_RPM_ALLOWED = MOTOR_MAX_RPM * MAX_RPM_RATIO

#define MOTOR_OPERATING_VOLTAGE 24          // motor's operating voltage (used to calculate max RPM)
#define MOTOR_POWER_MAX_VOLTAGE 24          // max voltage of the motor's power source (used to calculate max RPM)
#define MOTOR_POWER_MEASURED_VOLTAGE 22     // current voltage reading of the power connected to the motor (used for calibration)

#define COUNTS_PER_REV1 19308                 // wheel1 encoder's no of ticks per rev
#define COUNTS_PER_REV2 26823                 // wheel2 encoder's no of ticks per rev
#define COUNTS_PER_REV3 26742                 // wheel3 encoder's no of ticks per rev
#define COUNTS_PER_REV4 26837                 // wheel4 encoder's no of ticks per rev

#define WHEEL_DIAMETER 0.1               // wheel's diameter in meters
#define LR_WHEELS_DISTANCE 0.5            // distance between left and right wheels

#define PWM_BITS 10                         // PWM Resolution of the microcontroller
#define PWM_FREQUENCY 20000                 // PWM Frequency

#define SERVO_BITS 12                       // Servo PWM resolution
#define SERVO_FREQ 50                       // Servo PWM frequency

// INVERT ENCODER COUNTS
#define MOTOR1_ENCODER_INV false
#define MOTOR2_ENCODER_INV true
#define MOTOR3_ENCODER_INV false
#define MOTOR4_ENCODER_INV true

// INVERT MOTOR DIRECTIONS
#define MOTOR1_INV true
#define MOTOR2_INV true
#define MOTOR3_INV true
#define MOTOR4_INV true

// ENCODER PINS
#define MOTOR1_ENCODER_A 23
#define MOTOR1_ENCODER_B 4

#define MOTOR2_ENCODER_A 15
#define MOTOR2_ENCODER_B 19

#define MOTOR3_ENCODER_A 18
#define MOTOR3_ENCODER_B 5

#define MOTOR4_ENCODER_A 17
#define MOTOR4_ENCODER_B 16

// MOTOR PINS
#ifdef USE_GENERIC_2_IN_MOTOR_DRIVER
  #define MOTOR1_PWM 21 //Pin no 21 is not a PWM pin on Teensy 4.x, you can swap it with pin no 1 instead.
  #define MOTOR1_IN_A 20
  #define MOTOR1_IN_B 1

  #define MOTOR2_PWM 5
  #define MOTOR2_IN_A 6
  #define MOTOR2_IN_B 8

  #define MOTOR3_PWM 22
  #define MOTOR3_IN_A 23
  #define MOTOR3_IN_B 0

  #define MOTOR4_PWM 4
  #define MOTOR4_IN_A 3
  #define MOTOR4_IN_B 2

  #define PWM_MAX pow(2, PWM_BITS) - 1
  #define PWM_MIN -PWM_MAX
#endif

#ifdef USE_GENERIC_1_IN_MOTOR_DRIVER
  #define MOTOR1_PWM 32 //Pin no 21 is not a PWM pin on Teensy 4.x, you can use pin no 1 instead.
  #define MOTOR1_IN_A 2
  #define MOTOR1_IN_B -1 //DON'T TOUCH THIS! This is just a placeholder

  #define MOTOR2_PWM 25
  #define MOTOR2_IN_A 26
  #define MOTOR2_IN_B -1 //DON'T TOUCH THIS! This is just a placeholder

  #define MOTOR3_PWM 27
  #define MOTOR3_IN_A 14
  #define MOTOR3_IN_B -1 //DON'T TOUCH THIS! This is just a placeholder

  #define MOTOR4_PWM 12
  #define MOTOR4_IN_A 13
  #define MOTOR4_IN_B -1 //DON'T TOUCH THIS! This is just a placeholder

  #define PWM_MAX pow(2, PWM_BITS) - 1
  #define PWM_MIN -PWM_MAX
#endif
`

Initially, pin2 was supposed to be assigned pin33 but then I saw 33 was used for the battery. So i temporarily changed it to pin2. But the issues remained the same. micro-ROS connection was not complete.
When I changed the corresponding lines back to what it was initially, micro-ROS connection is successful after building.

hippo5329, 2024-05-31:
The lino_base_config.h is NOT used for esp32. Only esp32_config.h will be used. Use the original esp32_config.h. uncomment the line #define USE_MPU6050_IMU to enable mpu6050. Do not change any other lines in esp32_config.h. Follow the update workflow. Connect the mpu6050 and run test_sensors. Post the photos of your wiring.

hippo5329, 2024-05-31:
The pins assignment of esp32 is totally different from teensy. Do not use the wiring for teensy,

hippo5329, 2024-05-31:
Only 2 pwm pins mode is valid in current esp32_config.h . The other motor configurations is not valid. I will update generic 1 pin assignments later.

Leesanhang, 2024-05-31:
so actually, I should follow the wiring provided in the esp32_config file?

Leesanhang, 2024-05-31:
Correct me if im wrong, but the ESP32 has no Pin 0 right?
[image]

hippo5329, 2024-05-31:
I changed the motor interface pin assignments. It should work though I didn't test them yet. I ordered generic 1 similar to the cytron to test.

hippo5329, 2024-05-31:
You are right. esp32 30 pins version does not have d0. I push another update for motor pins. please pull again. Then perform the tests.

git pull

Leesanhang, 2024-06-02:
Hi again, I'm trying to follow your steps closely for now. However, I am facing an issue. Every time I shut down my Pi and come back the next day, micro-ROS will just magically not work. Do you maybe have an idea on why this is happening?

hippo5329, 2024-06-02:
What is the error message  from micro-ROS agent? Check 'ls /dev/ttyU* -l" to the the access. Which micro-SD you use? Write speed of the memory card is important.

Leesanhang, 2024-06-02:
I mean, it just doesnt connect fully as usual. However, I just realised what the problem is. 
There are a few parts to your steps: serial connection test, test sensors, test motors, and test acceleration right? 
Starting from test sensors, I uncomment the MPU6050 line. That's when the connection is not complete when I run `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600` . Is that normal?

Leesanhang, 2024-06-02:
I think the MPU might be the main problem in my case. 
**This is when MPU is commented out**, while doing https://github.com/hippo5329/linorobot2_hardware/wiki#connect-micro-ros-agent-again-and-check-imudata-topic
[image]

This is when MPU is uncommented, doing the same steps
[image]

Hope this helps with helping me find out what my problem is 🙏🏼🙏🏼🙏🏼

hippo5329, 2024-06-02:
Please uncomment the mpu6050 to enable IMU and run test_sensors. Press reset button to see the messages of i2cdetect.

hippo5329, 2024-06-02:
https://github.com/hippo5329/linorobot2_hardware/wiki#connect-the-imu-and-test_sensors

hippo5329, 2024-06-02:
If you have run this step, does the IMU reading correct in test_sensors?

hippo5329, 2024-06-02:
In your previous message, micro-ros2 agent can not connect because the esp32 stopped when the IMU init failed. You should see the LED blinks 3 times per sec.

Leesanhang, 2024-06-02:
I ran the test_sensors commands and this is what i got. 
[image]
The random characters are due to pressing the ESP reset button. 

Yes, the LED blinks thrice if micro-ROS connection failed.

hippo5329, 2024-06-02:
The IMU looks good in test_sensors.
Now, upload firmware and check the monitor . Press reset button to see the messages.

cd firmware
pio run -e esp32 -t upload
pio device monitor -e esp32

hippo5329, 2024-06-02:
The reading of IMU changed too much. Do not move the IMU and check test_sensors later. The reading should be close to 0,0,9.8 with normal stand.

We will check the firmware boot monitor message first.

Leesanhang, 2024-06-02:
This is when I `ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600`

[image]

I did not touch the IMU. Not sure why the values are fluctuating about those values.

[122 more replies]