hippo5329/linorobot2_hardware #19: built and uploaded successfully but loading forever on microros

iiZtZahran, 2024-06-07:
Hello, Hope yall have a nice a day
i`ve followed the steps to build for my esp32-wroom and builded successfully and i uploaded the hex on my esp which went successfull as well I run:
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600 
which gave me this:
[image]
i.e im using ubuntu mate 20.04 , esp32-wroom , mpu9250 which i configured in /custom/esp32_config.h 
and I tried to run the bringup using serial port /dev/ttyUSB0 (without connecting the lidar as you mentioned) and im getting this output as well which also continue running forever
[image]

hippo5329, 2024-06-07:
Please use copy & paste the screen output. It will be clear as my eyes are not so good with screen captured. :)

1. In your first message, the micro-ros is not connected.
2. In your second message, the rplidar driver got error. which model do you use?

So, we shall focus on the micro-ros connection first. Please comment out the IMU setting in esp32_config.h. This is step 2 in the workflow described in my wiki. Make sure the micro-ros is connected and the topics are checked.

https://github.com/hippo5329/linorobot2_hardware/wiki#test-micro-ros-agent-connection-before-wiring-and-building-the-robots---recommended

hippo5329, 2024-06-07:
After the micro-ros connection test success, proceed to step 3 in workflow, wire and enable the IMU. Add two 2k resistors pull up to 3.3 on the sda and scl lines. Run the test_sesnors.

https://github.com/hippo5329/linorobot2_hardware/wiki#connect-the-imu-and-test_sensors

Then follow the workflow step-by-step.

In step 8, robot bringup, you will need to use the long by-path device names.

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-lidar-on-the-robot-computer

iiZtZahran, 2024-06-09:
hello again, i followed your discussion in this issue solved the first problem and now my imu is working well and tested it with esp32 but , for the second part im still getting this issue:
**[rplidar_composition-5] [ERROR] [1717896175.095175069] [rplidar_composition]: Error, operation time out. RESULT_OPERATION_TIMEOUT!**
and this is my command:
**ros2 launch linorobot2_bringup bringup.launch.py \
    base_serial_port:=/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0-port0 \
    lidar_serial_port:=/dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.4:1.0-port0**  
reinstalled my linorobot2 and still getting this error also im on ros foxy and followed the wiki where you mentioned this https://github.com/hippo5329/linorobot2_hardware/wiki#install-linorobot2-on-ubuntu-2004-and-foxy

hippo5329, 2024-06-09:
What is your lidar model? What is the laser_type you used to install linorobot2?

For rplidar a1m8, use a1.
For rplidar a2m8, use a2m8.

hippo5329, 2024-06-09:
https://github.com/Slamtec/sllidar_ros2

Please try the launch to view lidar in rviz.

eg, for a1,
ros2 launch sllidar_ros2 view_sllidar_a1_launch.py serial_port:=< your by-path>

this->declare_parameter<std::string>("serial_port", "/dev/ttyUSB0");

I will check the bringup launch later.

iiZtZahran, 2024-06-09:
well i runned it and gave me this:
**[INFO] [launch]: All log files can be found below /home/mobile/ros_logs/2024-06-09-04-52-07-218499-MOBILE-4332
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sllidar_node-1]: process started with pid [4337]
[INFO] [rviz2-2]: process started with pid [4339]
[sllidar_node-1] [INFO] [1717897928.381700731] [sllidar_node]: SLLidar running on ROS2 package SLLidar.ROS2 SDK Version:1.0.1, SLLIDAR SDK Version:2.1.0
[rviz2-2] [INFO] [1717897930.769481769] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1717897930.770111339] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[sllidar_node-1] [ERROR] [1717897930.888376671] [sllidar_node]: Error, operation time out. SL_RESULT_OPERATION_TIMEOUT! 
[ERROR] [sllidar_node-1]: process has died [pid 4337, exit code 255, cmd '/home/mobile/linorobot2_ws/install/sllidar_ros2/lib/sllidar_ros2/sllidar_node --ros-args -r __node:=sllidar_node --params-file /tmp/launch_params_rsz4_6fs'].
[rviz2-2] [INFO] [1717897930.941909150] [rviz2]: Stereo is NOT SUPPORTED** 
and this is rviz:
[image]
[image]

hippo5329, 2024-06-09:
Do you use rplidar a1?

iiZtZahran, 2024-06-09:
yes im using a1m8 and specified a1 during installation also i know that the a1 works with 115200 baudrate and checked my usb port baudrate and was 115200 + i replaced the cables to ensure that its not hardware problem

hippo5329, 2024-06-09:
Let focus on the lidar, unlpug both esp32 and lidar.  Close all launch. Then plug the lidar. It will become ttyUSB0, the default port. Run the view launch.

hippo5329, 2024-06-09:
ros2 launch sllidar_ros2 view_sllidar_a1_launch.py

iiZtZahran, 2024-06-09:
same output:
[image]
and 
[INFO] [launch]: All log files can be found below /home/mobile/ros_logs/2024-06-09-05-03-06-300402-MOBILE-8107
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sllidar_node-1]: process started with pid [8109]
[INFO] [rviz2-2]: process started with pid [8111]
[sllidar_node-1] [INFO] [1717898586.685857693] [sllidar_node]: SLLidar running on ROS2 package SLLidar.ROS2 SDK Version:1.0.1, SLLIDAR SDK Version:2.1.0
[rviz2-2] [INFO] [1717898588.284313932] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1717898588.284992264] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[rviz2-2] [INFO] [1717898588.405806314] [rviz2]: Stereo is NOT SUPPORTED
[sllidar_node-1] [ERROR] [1717898589.195244969] [sllidar_node]: Error, operation time out. SL_RESULT_OPERATION_TIMEOUT! 
[ERROR] [sllidar_node-1]: process has died [pid 8109, exit code 255, cmd '/home/mobile/linorobot2_ws/install/sllidar_ros2/lib/sllidar_ros2/sllidar_node --ros-args -r __node:=sllidar_node --params-file /tmp/launch_params_a6hg_7j7'].

hippo5329, 2024-06-09:
Check it on a PC, ubuntu 22.04 Humble. or  ubuntu 20.04 Foxy.

Follow the readme to install, build and test view.

https://github.com/Slamtec/sllidar_ros2

hippo5329, 2024-06-09:
No linorobot2. Just slamtec along.

iiZtZahran, 2024-06-09:
same output as linorobot
[image]
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sllidar_node-1]: process started with pid [16147]
[INFO] [rviz2-2]: process started with pid [16149]
[sllidar_node-1] [INFO] [1717900157.064414454] [sllidar_node]: SLLidar running on ROS2 package SLLidar.ROS2 SDK Version:1.0.1, SLLIDAR SDK Version:2.1.0
[sllidar_node-1] [INFO] [1717900157.574401167] [sllidar_node]: SLLidar S/N: D9E999F6C9E59AD4C5E59CF7253B3414
[sllidar_node-1] [INFO] [1717900157.574556904] [sllidar_node]: Firmware Ver: 1.29
[sllidar_node-1] [INFO] [1717900157.574610143] [sllidar_node]: Hardware Rev: 7
[rviz2-2] [INFO] [1717900158.629105244] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1717900158.629738763] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[rviz2-2] [INFO] [1717900158.754220721] [rviz2]: Stereo is NOT SUPPORTED
[sllidar_node-1] [ERROR] [1717900159.574791862] [sllidar_node]: Error, cannot retrieve SLLidar health code: 80008002
[ERROR] [sllidar_node-1]: process has died [pid 16147, exit code 255, cmd '/home/mobile/ros2_ws/install/sllidar_ros2/lib/sllidar_ros2/sllidar_node --ros-args -r __node:=sllidar_node --params-file /tmp/launch_params_w9za907z'].

hippo5329, 2024-06-09:
If it doesn't work on PC, then it maybe problem of the lidar.

iiZtZahran, 2024-06-09:
I already tested before getting into the esp32 part and it worked well idk what happened, will try few solutions hoping it will work, thank you for your precious time I will an update once I reach out something

hippo5329, 2024-06-09:
Ok. Please note I forced updated the foxy branch on hippo5329/linorobot2. Now all ros2 branches use the same code. Please remove ~/linorobot2_ws and reinstall.

iiZtZahran, 2024-06-09:
hello again, I did some hardware checkups and changes and reinstalled from the main repo:
**https://github.com/linorobot/linorobot2**
I tested the lidar alone using **ros2 run rplidar_ros rplidar_composition --ros-args -p serial_port:=/dev/ttyUSB0 -p frame_id:=laser_frame** and it went well and showed up in rviz then I connected my esp32 and run the same bringup command but its still stuck at this point:
[rplidar_composition-5] [INFO] [1717928299.702748488] [rplidar_composition]: Firmware Ver: 1.29
[rplidar_composition-5] [INFO] [1717928299.702845507] [rplidar_composition]: Hardware Rev: 7
[rplidar_composition-5] [INFO] [1717928299.704417308] [rplidar_composition]: RPLidar health status : '0'
[rplidar_composition-5] [INFO] [1717928300.248326987] [rplidar_composition]: current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2, flip_x_axis 0

[image]

hippo5329, 2024-06-09:
This is another rplidar driver, which does not take serial port parameter.

        Node(
            condition=LaunchConfigurationEquals('sensor', 'rplidar'),
            name='rplidar_composition',
            package='rplidar_ros',
            executable='rplidar_composition',
            output='screen',
            remappings=[('scan', LaunchConfiguration('topic_name'))],
            parameters=[{
                'serial_port': '/dev/ttyUSB0',
                'serial_baudrate': 115200,  # A1 / A2
                'frame_id': LaunchConfiguration('frame_id'),
                'inverted': False,
                'angle_compensate': True,
            }],
        ),

hippo5329, 2024-06-09:
In later humble update, 

def launch_rplidar(context, *args, **kwargs):
    lidar_str = context.perform_substitution(LaunchConfiguration('sensor'))
    rplidar_sensors = [
        'a1',
        'a2',
        'a3',
        'c1',
        's1',
        's2',
        's3',
    ]
    
    if lidar_str in rplidar_sensors:
        launch_file = f'sllidar_{lidar_str}_launch.py'
        return [IncludeLaunchDescription(
            PythonLaunchDescriptionSource(PathJoinSubstitution(
                [FindPackageShare('sllidar_ros2'), 'launch', launch_file]
            )),
            launch_arguments={
                'serial_port': '/dev/rplidar', 
                'frame_id': LaunchConfiguration('frame_id'),
            }.items()   
        )]
    else:
        return []

So sllidar uses /dev/rplidar . Check udev rules. Can you find /dev/rplidar ?

hippo5329, 2024-06-09:
I changed the serial port in my fork.
                'serial_port': LaunchConfiguration('lidar_serial_port'),

hippo5329, 2024-06-09:
If there is /dev/rplidar, try lidar_serial_port:=/dev/rpliar in my fork.

hippo5329, 2024-06-09:
```
# set the udev rule , make the device_port be fixed by rplidar
#
KERNEL=="ttyUSB*", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", MODE:="0777", SYMLINK+="rplidar"
```

hippo5329, 2024-06-09:
I recalled that all the lidars use mostly the same usb id 10c4:ea60. So they all mixed up with esp32 usb port. That's why I added the lidar_serial_port parameter, which is not merged to linorobot/linorobot2 yet. I did send the PR.

For foxy, the old rplidar_composition drive might work better. I am not sure if the new sllidar driver works, though foxy is shown on their pages.

The linorobot/linorobot2 foxy branch does not have serial port parameter. I will add it when I have time.

hippo5329, 2024-06-10:
I  added lidar_serial_port parameter to hippo5329/linorobot2 foxy branch, forced updated. Please try out.

iiZtZahran, 2024-06-11:
hello again, I found out that my lidar work 1 time out of 5 times and when i try to do so it gives me the same error or error 800800 or 8008002 but when spamming the command it worked few times and i was able to bringup the robot successfully and everything was going well 
[image]
even tested creating a map and saving it successfully 
[image]
I will try the fixed branch tho and give you update soon

hippo5329, 2024-06-11:
I found this on rplidar error. "So I identified the problem was primarily due to the inadequate power from the usb ports. Once I switched to a external powered usb hub, it was solved."

hippo5329, 2024-06-11:
ubuntu@humble:~/ros2_ws$ ros2 launch sllidar_ros2 view_sllidar_a1_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-06-11-15-27-23-060774-humble-4039
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sllidar_node-1]: process started with pid [4040]
[INFO] [rviz2-2]: process started with pid [4042]
[sllidar_node-1] [INFO] [1718144843.457328743] [sllidar_node]: SLLidar running on ROS2 package SLLidar.ROS2 SDK Version:1.0.1, SLLIDAR SDK Version:2.1.0
[sllidar_node-1] [INFO] [1718144843.483534427] [sllidar_node]: SLLidar S/N: 858AED93C0EA98C7A0E69BF515544560
[sllidar_node-1] [INFO] [1718144843.483681046] [sllidar_node]: Firmware Ver: 1.29
[sllidar_node-1] [INFO] [1718144843.483709158] [sllidar_node]: Hardware Rev: 7
[sllidar_node-1] [INFO] [1718144843.485836082] [sllidar_node]: SLLidar health status : 0
[sllidar_node-1] [INFO] [1718144843.485888857] [sllidar_node]: SLLidar health status : OK.
[rviz2-2] [INFO] [1718144844.275075450] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1718144844.276877201] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-2] [INFO] [1718144844.495324669] [rviz2]: Stereo is NOT SUPPORTED
[sllidar_node-1] [ERROR] [1718144845.495982716] [sllidar_node]: Can not start scan: 80008000!

hippo5329, 2024-06-12:
So rplidar usb power is critical.

tested rplidar a1m8 on foxy pc.

bringup

ubuntu@mint-tiger:~$ ros2 launch linorobot2_bringup bringup.launch.py \
>     base_serial_port:=/dev/ttyUSB1 lidar_serial_port:=/dev/ttyUSB0
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-06-12-01-35-41-903361-mint-tiger-18450
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ekf_node-1]: process started with pid [18453]
[INFO] [micro_ros_agent-2]: process started with pid [18455]
[INFO] [joint_state_publisher-3]: process started with pid [18457]
[INFO] [robot_state_publisher-4]: process started with pid [18459]
[INFO] [rplidar_composition-5]: process started with pid [18462]
[micro_ros_agent-2] [1718156142.090901] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[micro_ros_agent-2] [1718156142.091073] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[rplidar_composition-5] [INFO] [1718156142.122183538] [rplidar_composition]: RPLIDAR running on ROS 2 package rplidar_ros. SDK Version: '1.12.0'
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base_link had 7 children
[robot_state_publisher-4] Link camera_link had 1 children
[robot_state_publisher-4] Link camera_depth_link had 0 children
[robot_state_publisher-4] Link front_caster_wheel_link had 0 children
[robot_state_publisher-4] Link imu_link had 0 children
[robot_state_publisher-4] Link laser had 0 children
[robot_state_publisher-4] Link left_wheel_link had 0 children
[robot_state_publisher-4] Link rear_caster_wheel_link had 0 children
[robot_state_publisher-4] Link right_wheel_link had 0 children
[robot_state_publisher-4] [INFO] [1718156142.136250728] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-4] [INFO] [1718156142.136356299] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1718156142.136369812] [robot_state_publisher]: got segment camera_depth_link
[robot_state_publisher-4] [INFO] [1718156142.136378387] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-4] [INFO] [1718156142.136386626] [robot_state_publisher]: got segment front_caster_wheel_link
[robot_state_publisher-4] [INFO] [1718156142.136395282] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-4] [INFO] [1718156142.136403042] [robot_state_publisher]: got segment laser
[robot_state_publisher-4] [INFO] [1718156142.136410710] [robot_state_publisher]: got segment left_wheel_link
[robot_state_publisher-4] [INFO] [1718156142.136418574] [robot_state_publisher]: got segment rear_caster_wheel_link
[robot_state_publisher-4] [INFO] [1718156142.136426435] [robot_state_publisher]: got segment right_wheel_link
[micro_ros_agent-2] [1718156142.302477] info     | Root.cpp           | create_client            | create                 | client_key: 0x0D65C523, session_id: 0x81
[micro_ros_agent-2] [1718156142.302513] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x0D65C523, address: 0
[micro_ros_agent-2] [1718156142.320683] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x0D65C523, participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.324658] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0D65C523, topic_id: 0x000(2), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.326522] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0D65C523, publisher_id: 0x000(3), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.329121] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0D65C523, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[micro_ros_agent-2] [1718156142.332579] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0D65C523, topic_id: 0x001(2), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.334490] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0D65C523, publisher_id: 0x001(3), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.336925] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0D65C523, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[micro_ros_agent-2] [1718156142.340485] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0D65C523, topic_id: 0x002(2), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.342470] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0D65C523, publisher_id: 0x002(3), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.344817] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0D65C523, datawriter_id: 0x002(5), publisher_id: 0x002(3)
[micro_ros_agent-2] [1718156142.347509] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0D65C523, topic_id: 0x003(2), participant_id: 0x000(1)
[micro_ros_agent-2] [1718156142.349492] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x0D65C523, subscriber_id: 0x000(4), participant_id: 0x000(1)
[... cut ...]
[rplidar_composition-5] [INFO] [1718156144.631959186] [rplidar_composition]: RPLidar health status : '0'
[rplidar_composition-5] [INFO] [1718156145.176588721] [rplidar_composition]: current scan mode: Sensitivity, max_distance: 12.0 m, Point number: 7.9K , angle_compensate: 2, flip_x_axis 0

ros2 topic hz /scan
WARNING: topic [/scan] does not appear to be published yet
average rate: 6.922
	min: 0.135s max: 0.148s std dev: 0.00552s window: 8
average rate: 6.938
	min: 0.134s max: 0.148s std dev: 0.00567s window: 15
average rate: 6.913
	min: 0.134s max: 0.148s std dev: 0.00537s window: 22

hippo5329, 2024-06-12:
slam
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-06-12-01-37-45-781303-mint-tiger-18669
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [async_slam_toolbox_node-1]: process started with pid [18671]
[INFO] [rviz2-2]: process started with pid [18673]
[async_slam_toolbox_node-1] [INFO] [1718156265.944474008] [slam_toolbox]: Node using stack size 40000000
[async_slam_toolbox_node-1] [INFO] [1718156265.969043758] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[async_slam_toolbox_node-1] [INFO] [1718156265.969482815] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[async_slam_toolbox_node-1] Registering sensor: [Custom Described Lidar]
[rviz2-2] [INFO] [1718156266.227360181] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1718156266.227530153] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1718156266.261902970] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] Parsing robot urdf xml string.
[rviz2-2] [INFO] [1718156270.978455640] [rviz2]: Trying to create a map of size 24 x 44 using 1 swatches
[rviz2-2] [ERROR] [1718156270.985567155] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
[rviz2-2] active samplers with a different type refer to the same texture image unit
[rviz2-2] [INFO] [1718156276.477405452] [rviz2]: Message Filter dropping message: frame 'laser' at time 1718156266.468 for reason 'Unknown'

hippo5329, 2024-06-12:
The laser scan displayed in rviz correctly.

iiZtZahran, 2024-06-12:
hello again, found out I was powering the raspberry with 15w power supply which was bit confusing since rpi 4 need 15w also I ordered an official adapter and my pcb will be offering the 15w as well I will be testing the whole system on both of  them the next few days when i finish the assembly, I really think since its power related problem as you say since everything worked fine few times.

hippo5329, 2024-06-12:
I think 15W power for rpi4 is fine. The problem is the power output voltage. Please check the 5V on the rpi4 with a multi-meter. There will be voltage drop from USB protection circuit. I used 5.25V power for rpi. You may check the VM on the tiny rplidar adapter.  You should use short USB cable. USB cables with fast charge support may have thicker wires. When the voltage on the rplidar is lower than 4.9V, it may not work well. This is a product design issue. I don't understand why their did not fix it. LD19 does not have such issue.

There is another issue in Slamtec sllidar_ros2 driver. The recent update has bug so most useful laser scan was discarded. I sent a pull request. I don't know when it will be merged. I am preparing another update on linorobot2 to resolve this.

hippo5329, 2024-06-12:
I updated hippo5329/linorobot2. Please follow the updated wiki to install linorobot2. The sllidar issue is fixed. There is no need to use the old rplidar_ros package.