hippo5329/linorobot2_hardware #28: Help me with the wheel and base_link connection issue when running the slam toolbox package.

HaiHai-17, 2024-11-10:
My Mecanum robot model, with hardware including a Raspberry Pi 4, ESP32, YDLidar, MPU6050, and OV5647 camera. Everything is working fine, including the laser, MPU, and encoder. However, when I start creating a map using the slam_toolbox package, I encounter the problem as shown in the image. Specifically, the wheels lose connection with the base link.
[image]

[image]

hippo5329, 2024-11-10:
Please launch bringup only. Check /scan and /imu/data hz in another terminal.

hippo5329, 2024-11-10:
bash install_linorobot2.bash mecanum ydlidar

Make sure you use by-path device names.

HaiHai-17, 2024-11-10:
The problem still hasn't been resolved, and I'm facing the same issue when running Gazebo :(

hippo5329, 2024-11-10:
As you have the robot. Ignore Gazebo for now.

Check /scan and /imu/data hz.

hippo5329, 2024-11-11:
Down/stop everything.

Follow this wiki to bringup the robot. Do not launch other staff.

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-lidar-on-the-robot-computer

Check the topics, echo, and find the hz rate.

https://github.com/hippo5329/linorobot2_hardware/wiki#check-topics

hippo5329, 2024-11-11:
Which ROS2 distro do you use? Are you running VM?

hippo5329, 2024-11-11:
Do you see four wheel links in joint state topic?

hippo5329, 2024-11-11:
Post the log message of bringup launch.

hippo5329, 2024-11-11:
You should have give more details in your configuration. Are you running ROS2 VM on Windows laptop?

You can not run VM on Windows to connect to ROS2 on rpi on robot. The RMW won't work without some discovery help such as discovery server. You can verify this issue by simple publisher/listener test to/from both sides.

You should use VNC server or remote desktop on rpi. Run all the ROS2 tasks on rpi.

HaiHai-17, 2024-11-11:
I'm currently using VMWare on Windows 11, Ubuntu 22.04, ROS2 Humble, and Gazebo for simulation. I'm using the git linorobot2 and when I launch Gazebo, the robot model displays normally. However, when I launch the slam_toolbox package to build a map, the wheels in RViz2 display an error as shown in the image, preventing the robot from moving around and building a map. The same error occurs when running on a real robot with a Raspberry Pi 4, Ubuntu 22.04, and ROS2 Humble.
[image]
[image]

HaiHai-17, 2024-11-11:
view tf2 
[image]

hippo5329, 2024-11-11:
Please check joint state msg as I asked.

"Do you see four wheel links in joint state topic?"

hippo5329, 2024-11-11:
I think you hacked too much. You should revert to my original code. The topic should be

```
ubuntu@cc187:~$ ros2 node list
/ekf_filter_node
/joint_state_publisher
/ld19
/pico
/robot_state_publisher
/transform_listener_impl_610ff0739b50

ubuntu@cc187:~$ ros2 topic list -t
/battery [sensor_msgs/msg/BatteryState]
/cmd_vel [geometry_msgs/msg/Twist]
/diagnostics [diagnostic_msgs/msg/DiagnosticArray]
/imu/data [sensor_msgs/msg/Imu]
/joint_states [sensor_msgs/msg/JointState]
/odom [nav_msgs/msg/Odometry]
/odom/unfiltered [nav_msgs/msg/Odometry]
/parameter_events [rcl_interfaces/msg/ParameterEvent]
/robot_description [std_msgs/msg/String]
/rosout [rcl_interfaces/msg/Log]
/scan [sensor_msgs/msg/LaserScan]
/set_pose [geometry_msgs/msg/PoseWithCovarianceStamped]
/tf [tf2_msgs/msg/TFMessage]
/tf_static [tf2_msgs/msg/TFMessage]
```

HaiHai-17, 2024-11-11:
Okay, I'll try it again from the beginning with your rolling version.

hippo5329, 2024-11-11:
It is important to check all the topics,

https://github.com/hippo5329/linorobot2_hardware/wiki#check-topics-1

closed 2024-11-24 by HaiHai-17