hippo5329/linorobot2_hardware #31: 2D Goal Pose not working in RVIZ

tbalatka, 2024-12-01:
I did a fresh install of Ubuntu 24.04 with ros2 Jazzy distro. After uploading the firmware I sent this command to start the robot: linorobot2_bringup bringup.launch.py micro_ros_transport:=udp4 micro_ros_port:=8888 lidar_transport:=udp_server lidar_server_port:=8889

I then start SLAM: ros2 launch linorobot2_navigation slam.launch.py rviz:=true

And I drive the robot around the room using: ros2 run teleop_twist_keyboard teleop_twist_keyboard

Issue: In RVIZ, when I attempt to use the 2D Goal Pose, the robot doesn't move.

BTW, the accuracy of the map generated is very good.

Here's my ros2 node list:
/behavior_server
/bt_navigator
/collision_monitor
/controller_server
/docking_server
/global_costmap/global_costmap
/launch_ros_23139
/lifecycle_manager_navigation
/local_costmap/local_costmap
/planner_server
/rviz2
/slam_toolbox
/smoother_server
/teleop_twist_keyboard
/transform_listener_impl_58380c84ccb0
/transform_listener_impl_5c92c5831470
/transform_listener_impl_61960e160160
/transform_listener_impl_62c675b05960
/transform_listener_impl_65193ab5cd20
/velocity_smoother
/waypoint_follower

hippo5329, 2024-12-01:
I am looking into this. Will share with you later.

hippo5329, 2024-12-01:
Looking into turtlebot4. We should follow the updates.

tbalatka, 2024-12-07:
I found this: https://github.com/turtlebot/turtlebot4/issues/501

Not sure whether it is helpful for us here.

hippo5329, 2024-12-30:
I followed the nav2 tutorials. The default nav2 configuration works with 2d goal pose slam. So it should be linorobot2 parameters.

To use default nav2, run each of the following in separate terminal.
```
ros2 launch linorobot2_bringup bringup.launch.py
ros2 launch nav2_bringup navigation_launch.py
ros2 launch slam_toolbox online_async_launch.py
ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/rviz/nav2_default_view.rviz

ros2 run nav2_map_server map_saver_cli -f ~/map
```

hippo5329, 2024-12-30:
We should specify nav2 params file with
```
ros2 launch nav2_bringup navigation_launch.py params_file:=<full/path/to/config/nav2_params.yaml>
```

It was in the robot description package config dir as the sambot tutorial.

hippo5329, 2024-12-30:
https://docs.nav2.org/setup_guides/footprint/setup_footprint.html#footprint-introduction

tbalatka, 2025-02-20:
Ok. I am making progress, I think. Question: where should I keep the Nav2 params yaml file? Should it be here: ...src/linobot2/linorobot2_navigation/config which already contains navigation_yaml, navigation_sim.yaml, slam.yaml ???

hippo5329, 2025-02-21:
You may save the nav2 params yaml file in that config directory, or the other directory you like.

Personally, I would keep only the linorobot2 robot bringup launch and skip the other parts such as nav/slam. I found it is better to stick to the nav2 guide, because it is maintained by the nav2 team.