hippo5329/linorobot2_hardware #33: Specific motor choice?

pitosalas, 2024-12-12:
As you know, I am using the Waveshare General Robot Driver board like you mention.

 I thought I would ask you here what specific motor you have used for a "beginner" robot? I am thinking about the general specs you mention, ~30cm ~2kg. So I think we need a 12v motor.

Which one did you use?

hippo5329, 2024-12-12:
Please search Amazon, Temu or AliExpress,12V DC encoder gear motor 150rpm wheel. You may also check robot chassis kit. The 6 pins sequence may be different from the board.

hippo5329, 2024-12-12:
https://github.com/hippo5329/linorobot2_hardware/wiki#sample-robot-build

These chassis were bought from Amazon. Motors with encoder are included.

hippo5329, 2024-12-12:
[image]

closed 2025-01-26 by pitosalas