hippo5329/linorobot2_hardware #36: IMU on Waveshare board

pitosalas, 2024-12-16:
Microros is talking to ROS2 successfully. Now I am on to the IMU.  Your instructions require me to do some soldering etc. I have a friend who can help me. But what stands in the way of relying on the IMU connection to the ESP32 that already exists on the board?

hippo5329, 2024-12-16:
There is no need to do soldering on the waveshare board. The IMU is connected. 

If you will have a robot computer such asrpi4/5 or mini pc/laptop on your robot, you should use serial transport and build with,
    pio run -e gendrv -t upload

If you do not have a robot computer, you will use wifi transport and build with,
    pio run -e gendrv_wifi -t upload

After the test_sensors works, you will need to update the motor/wheel parameters in the configuration file, config/custom/gendrv_config.h or gendrv_wifi_config.h .

hippo5329, 2024-12-16:
Only if you use wifi transport and LD19 lidar, you will use lidar forwarder and need to add a jumper wire to the on board header to connect the lidar rx data to io4. This can be done with a short regular female to female Dupont cable, which is popular in arduino projects.

pitosalas, 2024-12-16:
I have a raspi 4 directly attached to the wave share 40pin. 

Is the serial transport  for the esp32<->Pi connection (i.e. for pio etc.)? I started by usbc to usba cable between the wave share and the raspi. Maybe I don't need that?

hippo5329, 2024-12-16:
Only power and i2c are connected on the 40 pins header. You may check the schematic. There is serial connection. You should connect the usb-c cable on wareshare board to pi. You will use serial transport and connect your lidar to pi directly.

pitosalas, 2024-12-16:
Thanks. So  linorobot_hardware/config/custom/gendrv.h takes the place of linorobot_hardware/config/config.h

What is gender?

hippo5329, 2024-12-16:
gendrv is Waveshare General Driver for robot. :)
The imu driver, motor drivers and encoders pins on waveshare board are configured. You will need to change the motor/wheel parameters though.

pitosalas, 2024-12-16:
So wherever the pio package name esp32 shows up in your instructions I should say gendrv correct?

Pito

hippo5329, 2024-12-16:
Yes. Please use gendrv instead of esp32 as platformio env. We use env to specify board. esp32 is generic.

https://github.com/hippo5329/linorobot2_hardware/wiki#configuration-examples

I would create a custom env/config for each robot build. You may use gendrv as a template and make a custom env/config later.

https://github.com/hippo5329/linorobot2_hardware/wiki#custom-configuration-file---recommended

closed 2025-01-24 by pitosalas