hippo5329/linorobot2_hardware #37: Lidar question

pitosalas, 2024-12-16:
I now have waveshare + Raspi 4 + IMU working correctly, thanks!

1) Is [WayPonDEV FHL-LD19 Plus 2D Lidar Sensor]( https://www.amazon.com/WayPonDEV-FHL-LD19-Plus-Lidar-Sensor/dp/B0DP4HQ8VB/ref=sr_1_3?crid=2I1MSCRME0QPU&dib=eyJ2IjoiMSJ9.iT6iQpGFyjZe3Veiq29cZfQSgBbgIcl-yD-5XHKxYfvXrh2MViSx3pRrUMg6sv4awhpN5VBCIzvAu-ksoj90fNsRZYAhTVECNVrz9tiw-6ldi6nDJmx4blVvIO3WuRV1ky8W-Gc3hOKIrWnGB7BpYj9ybv30nU4R-1UurHUAww9iOcwH-iffIzn8grioLXHC.SGPFkgmJSQABoj50pXbxQZy0JDV9H43D5jVM_DLQ3p0&dib_tag=se&keywords=lidar+ld19&qid=1734383270&s=industrial&sprefix=lidar+ld19%2Cindustrial%2C94&sr=1-3 ) the same as [DTOF LD19 Laser LiDAR Sensor Kit](https://www.dfrobot.com/product-2610.html)?

2) Can I somehow connect the Lidar directly without the USB connector? Either to the Pi or to the gendev?

3) I have an LDS-02 Lidar (from Robotis). It also has USB connector. How hard is it to make it work with gendev and your fork?

Thank you!

hippo5329, 2024-12-16:
1. They are the same. Some named it as D300, too.
2. Yes, you can. LD19 is simple, 5V GND, RX data to pi. You will need to assign lidar serial port in your linorobot2 bringup launch. esp32 might not have the resources to handle the lidar messages. In wifi setup, I forwarded the lidar data to udp without processing. However, I would suggest USB connections for both the waveshare and lidar as the initial tests. I believe USB serial will be easier than hardware serial ports on pi. It will be more flexible when you want to try out jetson or mini pc/laptop.
3. I think it will be easy. The lidar has nothing to do with micro-ROS/waveshare. I think LDS-02 lidar ROS2 driver should be available with turtlebot3. I will check more details later. The laser launch in the linorobot2 bringup will need some mod.

Please note both waveshare and lidar might use similar device id/names. udev rules might be useless. We will need to specify the ports with by-path names.

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-lidar-on-the-robot-computer

hippo5329, 2024-12-16:
This is the lidar launch file, which start the lidar driver.

https://github.com/linorobot/linorobot2/blob/humble/linorobot2_bringup/launch/lasers.launch.py

Before connect lidar to linorbot2. You should run lidar view test program to see the lidar scan on rviz2.

LDS-02 = LD08 driver

https://github.com/ROBOTIS-GIT/ld08_driver

hippo5329, 2024-12-16:
I am able to build the ld08 driver with humble branch. But there is no view launch. You may try to view the  scan on rviz2. The serial port assignment will be an problem as it uses cp2102 which is the same as waveshare. We will need to modify the driver src/main.cpp to use our lidar port.

closed 2025-01-24 by pitosalas