hippo5329/linorobot2_hardware #44: Pi not talking to ESP32

pitosalas, 2025-01-21:
Hi!

I have a problem that has cropped up. It worked before and now it is failing. I suspect a baud rate problem but don't know how to troubleshoot it.

```
[dome:~/linorobot2_hardware/firmware] ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0
[1737425652.773743] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1737425652.775324] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

And it hangs there, I suspect, waiting to set up a session. 

When I list the serial ports I get this:

```
Device: /dev/ttyUSB0, Name: ttyUSB0, Description: CP2102N USB to UART Bridge Controller, HWID: USB VID:PID=10C4:EA60 SER=eec5fe24071cef119ca199d94909ffd0 LOCATION=1-1.2
Device: /dev/ttyAMA0, Name: ttyAMA0, Description: ttyAMA0, HWID: fe201000.serial
```

Where I think that USB0 is the ESP32 (on the Waveshare general board). 

But trial and error is not getting me there. Can you point me in the right direction?

---
p.s. to list the serial ports my code looks like this:

```
from serial.tools import list_ports
ports = list_ports.comports()
for port in ports:
    print(f"Device: {port.device}, Name: {port.name}, Description: {port.description}, HWID: {port.hwid}")

```

hippo5329, 2025-01-21:
You will need to specify the baudrate as 921600.

  ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600

hippo5329, 2025-01-21:
Please note that we will need to specify by-path device names for esp32 and lidar. They use the same cp2102 USB serial chips.

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-lidar-on-the-robot-computer

The waveshare board has magnetometer. So we will use madgwick to fuse imu and mag.

Like this, but your devices path will be different.
```
ros2 launch linorobot2_bringup bringup.launch.py  madgwick:=true orientation_stddev:=0.01 \
    base_serial_port:=/dev/serial/by-path/pci-0000:00:14.0-usb-0:2.4.2:1.0-port0 \
    lidar_serial_port:=/dev/serial/by-path/pci-0000:00:14.0-usb-0:2.4.4:1.0-port0
```

pitosalas, 2025-01-21:
> The micro-ros connection works in your test. The "Waiting for robot_description to be published" message is normal. The missing /cmd_vel is strange.
>
> You may connect waveshare board to your PC. You may use container if your os host is not ubuntu. My PC runs fedora bluefin and I use distrobox/podman to run ubuntu/ros2.
>
> I would suggest test on PC or Pi,
>
> https://github.com/hippo5329/linorobot2_hardware/wiki#test-micro-ros-agent-connection-before-wiring-and-building-the-robots---recommended
>
> cd linorobot2_hardware/firmware
> pio run -e gendrv -t upload
> ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600

Thanks. I’ve got something weird happening…

[dome:~/linorobot2_hardware/firmware] ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600
[1737493510.097116] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1737493510.098831] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1737493510.562888] info     | Root.cpp           | create_client            | create                 | client_key: 0x60E7DF1D, session_id: 0x81
[1737493510.564805] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x60E7DF1D, address: 0
[1737493510.675391] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x60E7DF1D, participant_id: 0x000(1)
[1737493510.679946] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x60E7DF1D, topic_id: 0x000(2), participant_id: 0x000(1)
[1737493510.683330] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x60E7DF1D, publisher_id: 0x000(3), participant_id: 0x000(1)
[1737493510.687787] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x60E7DF1D, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1737493510.692078] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x60E7DF1D, topic_id: 0x001(2), participant_id: 0x000(1)
[1737493510.695717] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x60E7DF1D, publisher_id: 0x001(3), participant_id: 0x000(1)
[1737493510.700153] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x60E7DF1D, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1737493510.704446] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x60E7DF1D, topic_id: 0x002(2), participant_id: 0x000(1)
[1737493510.707987] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x60E7DF1D, publisher_id: 0x002(3), participant_id: 0x000(1)
[1737493510.712768] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x60E7DF1D, datawriter_id: 0x002(5), publisher_id: 0x002(3)
[1737493510.717281] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x60E7DF1D, topic_id: 0x003(2), participant_id: 0x000(1)
[1737493510.720825] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x60E7DF1D, publisher_id: 0x003(3), participant_id: 0x000(1)
[1737493510.725259] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x60E7DF1D, datawriter_id: 0x003(5), publisher_id: 0x003(3)
[1737493510.729465] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x60E7DF1D, topic_id: 0x004(2), participant_id: 0x000(1)
[1737493510.732962] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x60E7DF1D, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1737493510.737370] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x60E7DF1D, datareader_id: 0x000(6), subscriber_id: 0x000(4)

But then

Ros2 node list gives no nodes and ros2 topic list -t gives only:

/parameter_events [rcl_interfaces/msg/ParameterEvent]
/rosout [rcl_interfaces/msg/Log]

So this is even weirder than no /cmd_vel

Any idea?

hippo5329, 2025-01-21:
The agent cannot communicate with the other nodes. What is your system configuration? Native or container?

If you run ros2 jazzy, do not use cyclonedds. Keep the default fastrtsp.

If you run ros2 humble, check host ip with "ip addr". Multiple network interfaces or ip addresses may cause problems.

hippo5329, 2025-01-21:
Or, return to basic micro-ros example.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

pitosalas, 2025-01-21:
I thought it might be something like that.

I am running on a raspberry Pi natively. Only one computer (the Pi) is involved.

I am using humble.

[dome:~/linorobot2_ws] printenv | grep ROS
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DOMAIN_ID=5
ROS2_WS=linorobot2_ws
ROS_LOCALHOST_ONLY=0
ROS_DISTRO=humble

Pito

hippo5329, 2025-01-21:
Try not set ROS_DOMAIN_ID. Use the default 0. Reboot..

hippo5329, 2025-01-21:
ROS_DOMAIN_ID should be configured in the micro-ros client, not the agent. Though it is not trivial..

pitosalas, 2025-01-21:
Removing it fixed it.

But for my info, which one is the client and which one is the agent?

Thanks

Pito

hippo5329, 2025-01-22:
esp32 is the client.
micro-ros agent is the agent.

hippo5329, 2025-01-22:
The ROS_DOMAIN_ID is said to be configured in a meta file for middleware. But I didn't see a correct answer yet.

pitosalas, 2025-01-22:
> micro-ros agent is the agent.
And it runs on the Pi. Yes?

I have a new problem which I will troubleshoot myself. But once I send it a single cmd_vel (with teleop) it starts moving and nothing will stop it. I suspect some of my parameters are wrong. 

Thanks for your help!

pitosalas, 2025-01-22:
Do you happen to have a script that will simply display the encoder counts? I want to empirically determine encoder tics per rotation.

hippo5329, 2025-01-22:
I use test_motors and tachometer to verify encoder count and rpm.

https://github.com/hippo5329/linorobot2_hardware/wiki#test-the-motors-and-sensors

closed 2025-01-22 by pitosalas