hippo5329/linorobot2_hardware #45: test_motors gives incorrect rad/s

PaulBouchier, 2025-01-21:
I was running test_motors on my differential drive robot and noticed it prints a rad/sec value which is two times what my hand-calculation says it should be based on one wheel spinning and one wheel stopped.

MOTOR1 FWD RPM    158.4      0.0      0.0      0.0
MOTOR1 SPEED   0.66 m/s   8.14 rad/s STOP  0.087 m
The wheels are 80mm diameter and 163mm apart.

Looking at the code, I think it is misusing the Kinematics class, and shouldn't be using that class at all.

The purpose of test_motors is to print the RPM and velocity of each wheel as it runs them in turn. Since only one motor runs at a time, and the focus is on the running motor, it is irrelevant what the platform would be doing, so the use of Kinematics is inappropriate. 

Wheel speed is computed in Kinematics as 
vel.linear_x = average_rps_x * wheel_circumference_; // m/s
and this IMHO is the calculation that should be done in test_motors. It only gives the right answer for the single running wheel because test_motors passes max_rpm in as if both the left and right motors were running, so the wheel speed average works out. It is incorrect if the platform was put on the ground and allowed to spin on the stationary wheel. IOW knowledge of how Kinematics is implemented is being used by test_motors, which works but is poor practice.

One would think, since test_motors is printing wheel speeds, that rad/s would be the angular velocity of the wheel itself, which  would be RPM/60 * 2 * pi. This doesn't seem like an interesting parameter, and IMHO it should be removed from the print. However, it passes wheel RPM to left and negative wheel RPM to right inputs of Kinematics. Kinematics computes what the angular speed of the whole platform would be if the left and right wheels were spinning in opposite directions, but they aren't - only one wheel is spinning. And in any case, platform angular velocity is not interesting for the purpose of this test.

I think the rad/s part of the print should be removed, and linear velocity should be computed in test_motors using wheel_circumference_ and Kinematics should not be used - it's relevant to the entire platform, not one wheel. 

Let me know if you agree and I can propose a PR.

Thanks for your great work in this repo and wiki!

hippo5329, 2025-01-21:
You are right. The RAD is incorrect and should be removed. Please proceed to PR. Thanks.

PaulBouchier, 2025-06-08:
Closed by PR #46

closed 2025-06-08 by PaulBouchier