hippo5329/linorobot2_hardware #47: [joint_state_publisher]: Waiting for robot_description to be published on the robot_

pitosalas, 2025-01-22:
```
The bringup launch will pause and wait for initial pose, which will be set by navigation lauch.

[joint_state_publisher]: Waiting for robot_description to be published on the robot_
```
I don't understand what this is telling me. What is an initial pose in this context?

hippo5329, 2025-01-22:
When we run nav2 navigation with a map, an initial pose/position will be given by user as the estimated position of robot in the map. Then nav2 stack AMCL will locate the robot during navigation.

The "Waiting for robot_description.." log message sent when joint state publisher listens on the robot_description topic instead of an URDF file.

https://github.com/ros/joint_state_publisher/blob/ros2/joint_state_publisher/joint_state_publisher/joint_state_publisher.py#L403

pitosalas, 2025-01-22:
I get that. But does that mean that the urdf for the robot is not displayed in rviz2 until then? Because in my application I often am not using a map. I don’t understand that dependency…

hippo5329, 2025-01-22:
That message should be taken as INFO. It is not WARNING or ERROR. The robot_description is already published. And the joint_state_publisher is running. There is no real "waiting".

closed 2025-01-22 by pitosalas