hippo5329/linorobot2_hardware #48: Counts Per Rev - Manual

pitosalas, 2025-01-22:
I think you said you have a device to measure actual RPM. I don't have that. So instead I wrote a tiny program to print out motor1_encoder.read() and motor2_encoder.read(). 

I am taking those to be the counts. I manually turned the wheel for ten full rotations and it gave me a count of  9854. So I therefore set the COUNTS_PER REV as follows.

#define COUNTS_PER_REV1 985                 // wheel1 encoder's no of ticks per rev

But the motor is turning very slowly which makes me believe that the exact meaning of counts_per_rev is a little different. 

Can you clarify? Thanks

hippo5329, 2025-01-22:
The two motor tests have different purpose.

test_motor is used to verify motor operation without load without touching the ground. It is used to check rpm and count per rev.

test_acc is used to find max vel and acc with full load on the ground after test_motor done.

So you should check test_motor first. And verify rpm with a tachometer.

hippo5329, 2025-01-22:
https://www.amazon.com/gp/product/B0BYPC2Y5V/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

hippo5329, 2025-01-22:
The count per revolution depends on your gear motor. It will be different from mine. You should check the datasheet of your gear motor.

pitosalas, 2025-01-23:
Unfortunately it’s a motor I had in a box that was left over and it has no markings on it nor a data sheet. 

I ordered the tachometer!

Pito

closed 2025-01-24 by pitosalas