hippo5329/linorobot2_hardware #49: Gendrv Board, ESP32,  Upload Error

pitosalas, 2025-01-24:
I am able to upload Firmware, run ROS2, upload it again, usually. But then from time to time I get this error. The only thing that seems to clear it is a full power cycle.

* reset button on ESP32 didn't work
* unplugging and plugging the USB connector didn't work
* trying again didn't work

Have you seen this, and any thoughts?

```
Using manually specified: /dev/ttyUSB0
Uploading .pio/build/gendrv/firmware.bin
esptool.py v4.5.1
Serial port /dev/ttyUSB0
Connecting......

A serial exception error occurred: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 1
================================================ [FAILED] Took 84.03 seconds ================================================

Environment    Status    Duration
-------------  --------  ------------
gendrv         FAILED    00:01:24.027
```

hippo5329, 2025-01-24:
Both micro-ROS agent and the esp32 upload protocol use the same serial port. So you have to close micro-ROS agent to perform upload.

hippo5329, 2025-01-24:
You cannot use serial monitor and upload at the same time.

pitosalas, 2025-01-24:
I thought I did… I killed ros2 launch linorobot2_bringup bringup…

Is there something more that I have to kill

hippo5329, 2025-01-24:
That should be enough. Please also make sure /dev/ttyUSB0 exist.

Another tip,

https://github.com/hippo5329/linorobot2_hardware/wiki#reset-and-power-onoff-usb-devices

pitosalas, 2025-01-26:
I think I figured out that if anything is running that uses ros2 (e.g. Rviz, rat, whatever) then it will keep micro_ros alive. What I’ve done is to try ros2 node list and if it shows nothing then I think micro_ros is not running.

closed 2025-01-26 by pitosalas