hippo5329/linorobot2_hardware #50: Gendrv board, orientation of IMU

pitosalas, 2025-01-24:
I don't know if you have a gendrv board and whether you test with it....

How is the IMU oriented on the board?

For example, I see the z axis acceleration as -9.x you see it as +9.x

I am still building the robot so I can relocate the board if needed.

hippo5329, 2025-01-24:
I have several waveshare boards. Please check the third picture in my [build photos](https://github.com/hippo5329/linorobot2_hardware/wiki#sample-robot-build) .

It is designed to face downwards. Pi will also face downwards and plug on the waveshare board. Pi will be mounted higher than the waveshare board. There is a video on their UGV robot.

https://www.youtube.com/watch?v=vB00DBXsZtY

hippo5329, 2025-01-24:
The wiki is not clear on these details. I will improve it later.

hippo5329, 2025-01-24:
https://www.waveshare.com/product/robotics/mobile-robots/raspberry-pi-robots/ugv-beast-ros2-kit.htm

pitosalas, 2025-01-24:
I see. I have mine mounted on top, so upside down from yours.

￼

And what about the direction of “X”?

hippo5329, 2025-01-24:
X is the opposite side of USB headers.

closed 2025-01-24 by pitosalas