hippo5329/linorobot2_hardware #51: Best Practice for a custom URDF

pitosalas, 2025-01-25:
I need to tweak some things in the URDF.

I could modify the basic linorobot2 package but then I will lose touch with your upstream version.

I could figure out how to use the custom_robot argument. But I am not quite sure yet whether that means a whole custom bringup or just a custom linorobot2_description or a custom urdf.

I could fork the GitHub repo

Do you have pros and cons or a recommendation?
'
Pito

hippo5329, 2025-01-25:
Please check the wiki, https://github.com/hippo5329/linorobot2_hardware/wiki#configure-urdf-in-linorobot2

There is no change in the urdf from the upstream linorobot.

You may use the default for initial test. Try teleop via keyboard. You may modify urdf later.

pitosalas, 2025-01-25:
Thanks… It IS working. 

But the IMU is upside down so I have to change it’s urdf, yes?

Also, separate but related, the robot model does not display in rviz2. Even though it does display in foxglove studio…

Pito

pitosalas, 2025-01-25:
> Also, separate but related, the robot model does not display in rviz2. Even though it does display in foxglove studio…

I fixed that

But the Lidar is way up in the sky. Another correction needed in urdf IMO

Pito

hippo5329, 2025-01-25:
Yes, you will have to modify urdf for these.

hippo5329, 2025-01-25:
I didn't pay attention to robot model. Maybe rviz configuration issues.

closed 2025-01-26 by pitosalas