hippo5329/linorobot2_hardware #53: Gendrv Issue

pitosalas, 2025-01-27:
I am having trouble getting the Waveshare gendrv board fully working. This issue looks long but it's just because I put all the details in.

I think I might be misunderstanding some detail in your instructions because of how you interleave instructions for gendrv with all the others. 

I have followed your instructions carefully and it is mostly working.  

## My gendrv_config.h looks like this:

```

// Copyright (c) 2021 Juan Miguel Jimeno
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#ifndef GENDRV_CONFIG_H
#define GENDRV_CONFIG_H

#define LED_PIN LED_BUILTIN //used for debugging status

//uncomment the base you're building
#define LINO_BASE DIFFERENTIAL_DRIVE       // 2WD and Tracked robot w/ 2 motors
// #define LINO_BASE SKID_STEER            // 4WD robot
// #define LINO_BASE MECANUM               // Mecanum drive robot

//uncomment the motor driver you're using
// #define USE_GENERIC_2_IN_MOTOR_DRIVER      // Motor drivers with 2 Direction Pins(INA, INB) and 1 PWM(ENABLE) pin ie. L298, L293, VNH5019
// #define USE_GENERIC_1_IN_MOTOR_DRIVER   // Motor drivers with 1 Direction Pin(INA) and 1 PWM(ENABLE) pin.
#define USE_BTS7960_MOTOR_DRIVER        // BTS7970 Motor Driver
// #define USE_ESC_MOTOR_DRIVER            // Motor ESC for brushless motors

//uncomment the IMU you're using
// #define USE_GY85_IMU
// #define USE_MPU6050_IMU
// #define USE_MPU9150_IMU
// #define USE_MPU9250_IMU
#define USE_QMI8658_IMU
// #define USE_HMC5883L_MAG
// #define USE_AK8963_MAG
// #define USE_AK8975_MAG
#define USE_AK09918_MAG
// #define USE_QMC5883L_MAG

#define MAG_BIAS { 1.32662e-05, -5.43438e-05, 1.06151e-05 } //magnetometer bias
// #define IMU_TWEAK {}
// #define MAG_TWEAK {}

#define K_P 0.6                             // P constant
#define K_I 0.8                             // I constant
#define K_D 0.5                             // D constant

// #define K_P 3.0                            // P constant
// #define K_I 0.0                             // I constant
// #define K_D 0.0                             // D constant

#define ACCEL_COV { 0.01, 0.01, 0.01 }
#define GYRO_COV { 0.001, 0.001, 0.001 }
#define ORI_COV { 0.01, 0.01, 0.01 }
#define MAG_COV { 1e-12, 1e-12, 1e-12 }
#define POSE_COV { 0.001, 0.001, 0.001, 0.001, 0.001, 0.001 }
#define TWIST_COV { 0.001, 0.001, 0.001, 0.003, 0.003, 0.003 }
[... 436 lines cut ...]
[robot_state_publisher-5] [INFO] [1737941559.019033586] [robot_state_publisher]: got segment right_wheel_link
[robot_state_publisher-5] [INFO] [1737941559.019060771] [robot_state_publisher]: got segment sonar_link
[micro_ros_agent-3] [1737941559.172003] info     | Root.cpp           | create_client            | create                 | client_key: 0x2CF3503A, session_id: 0x81
[micro_ros_agent-3] [1737941559.174546] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2CF3503A, address: 0
[micro_ros_agent-3] [1737941559.236455] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x2CF3503A, participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.241098] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2CF3503A, topic_id: 0x000(2), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.244672] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x2CF3503A, publisher_id: 0x000(3), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.249848] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x2CF3503A, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[micro_ros_agent-3] [1737941559.254031] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2CF3503A, topic_id: 0x001(2), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.257495] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x2CF3503A, publisher_id: 0x001(3), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.263481] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x2CF3503A, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[micro_ros_agent-3] [1737941559.267806] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2CF3503A, topic_id: 0x002(2), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.271348] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x2CF3503A, publisher_id: 0x002(3), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.277392] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x2CF3503A, datawriter_id: 0x002(5), publisher_id: 0x002(3)
[micro_ros_agent-3] [1737941559.283414] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2CF3503A, topic_id: 0x003(2), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.287764] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x2CF3503A, publisher_id: 0x003(3), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.296514] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x2CF3503A, datawriter_id: 0x003(5), publisher_id: 0x003(3)
[micro_ros_agent-3] [1737941559.300748] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x2CF3503A, topic_id: 0x004(2), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.304181] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x2CF3503A, subscriber_id: 0x000(4), participant_id: 0x000(1)
[micro_ros_agent-3] [1737941559.310181] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x2CF3503A, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[imu_filter_madgwick_node-1] [INFO] [1737941559.347430177] [madgwick_filter_node]: First pair of IMU and magnetometer messages received.
[joint_state_publisher-4] [INFO] [1737941560.837289555] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
```

## After this I have a few mysteries. I think they are both due to gendrv and the magnetometer feature of the imu.

hippo5329, 2025-01-27:
The problem came from the madgwick package, which used "fixed_frame"(default odom) as parent . But it should not be odom in our cases.

hippo5329, 2025-01-27:
I think we can set "publish_tf" to false, because we do not use wadgwick package to publish tf. We use robot_state_publisher to publish tf. So the tf publisher from madgwick should be disabled.

hippo5329, 2025-01-27:
I will prepare a patch on this later.

pitosalas, 2025-01-27:
I am pretty sure I tried that and it created other errors. Pito Salas Computer ScienceBrandeis Univeristy

pitosalas, 2025-01-28:
Hi Thomas,Any news on this patch?Pito Salas Computer ScienceBrandeis Univeristy

hippo5329, 2025-01-29:
Please check hippo5329/linorobot2 no-tf branch. I am traveling and do not have robots to test.

pitosalas, 2025-01-29:
Thank you. Unfortunately it didn't fix the problem. I am still seeing the imu tf parent as odom which it should not be. Here is a short capture to show what I am seeing…

Thanks for all you do!

https://share.zight.com/KouRAWDX

￼

hippo5329, 2025-02-09:
I updated hippo5329/linorobot2. And checked via view_frame. The imu_link's parent is base_footprint now.

closed 2025-02-14 by pitosalas