hippo5329/linorobot2_hardware #55: Documentation suggestion

pitosalas, 2025-02-14:
In the highly useful and detailed documentation in the readme for this repo, it is sometimes/often unclear whether you are referring to the onboard raspi or the remote computer. For example: 

[image]

This ambiguity occurs in various places and can be confusing.

hippo5329, 2025-02-18:
Only serial micro-ROS agent and serial lidar driver need to be installed on the robot computer. The other packages can be installed on either robot computer or remote computer.

For micro-ROS wifi transport and LD19 wifi forwarder, there is no robot computer. All ROS2 packages are installed on the remote computer.

Personally, I prefer remote computers/servers because they are easier to use and much faster than raspi4/5. But this really depends on applications.

pitosalas, 2025-02-24:
"Personally, I prefer remote computers/servers because they are easier to use and much faster than raspi4/5. But this really depends on applications."

Interesting... I do like that idea. But I wonder about performance of the ESP32 microcontroller as well as the bandwidth of the wifi. Does your linorobot2 fork support that setup? And what about Lidar and Camera which are traditionally over USB to the pi?

hippo5329, 2025-02-25:
Esp32 has on-chip wifi interface, which is much faster than the SPI wifi on the pico-w and others. The sustained bandwidth is around 10Mbit/s.

I wrote a LD19 forwarder to send the lidar raw data to UDP server on host. I use my linorobot2/linorobot2_hardware forks. I have only one Pi 5. Most of my robots do not have a robot computer on them. Rplidar could be used this way, but I do not have the interest to work on them.

I will use a separate esp32-cam for video. I am working on this. I am learning opencv, yolo and openvino.

closed 2025-02-26 by pitosalas