hippo5329/linorobot2_hardware #56: Hz for odom is just 38

pitosalas, 2025-02-14:
I don't think this matters, but your doc suggests it should be around 50. What is the significance of a slightly lower hz?

```
[dome:~] ros2 topic hz /odom
average rate: 38.448
	min: 0.019s max: 0.040s std dev: 0.00909s window: 40
average rate: 38.343
	min: 0.019s max: 0.041s std dev: 0.00920s window: 79
average rate: 38.307
	min: 0.019s max: 0.041s std dev: 0.00921s window: 118
```

hippo5329, 2025-02-15:
It is fine. I am not able to resolve this. I guess the imu mag driver caused the issue.

pitosalas, 2025-02-15:
The mag seems to be very inaccurate (or random) indoors. Is it helping at all or should I just pretend it does not exist?

hippo5329, 2025-02-15:
Yes. It is inaccurate. It is interfered by the lidar, motors and others. So I set the covariance high, 0.01. It does help to give geographic orientation. 

ros2 launch linorobot2_bringup bringup.launch.py madgwick:=true orientation_stddev:=0.01

You may set the covariance higher or disable the mag by comment out,

https://github.com/hippo5329/linorobot2_hardware/blob/master/config/custom/gendrv_config.h#L40

closed 2025-02-26 by pitosalas