hippo5329/linorobot2_hardware #57: Gendrv - IMU and magnetometer calibration

pitosalas, 2025-02-14:
I have calibrated the magnetometer according to your documentation and rebuilt the firmware.

On the IMU for gendrv, you don't seem to have special documentation (and neither does the upstream code). So I used the ROS2 imu_calib package. I ran the calibration and I now have the imu_calibration.yaml. 

Now I have to run the apply_calib node. I am surprised it is not invoked in the standard bringup launch file provide by linorobot2. Maybe I misunderstand when it is supposed to be run. 

My Imu is upside down, that is +z of the chip is pointing down. Should I change the way I hold the robot while calibrating so that when it asks to point +z up I hold the robot upside down?

By the way, does imu_calib look at the urdf to determine the orientation of the imu board? 

When / how do I run apply_calib?

Thanks!

hippo5329, 2025-02-15:
I use robot_calibration package.

https://github.com/hippo5329/linorobot2_hardware/wiki#magnetometer-calibration

I don't know if it use URDF. We will need to look into the code.

The calibration should be done in-place with everything installed. So temporary reverse the board is not good.

pitosalas, 2025-02-15:
I used that for the magnetometer too. But what did you use to calibrate the IMU? I used ROS2's imu_calib. It generates the x, y and z corrections in a text file. But they have to be applied, which requires that the apply_calib node. I don't see that referenced anywhere.

What did you use to calibrate the IMU?

hippo5329, 2025-02-15:
Most IMU have self-calibration during initialization, such as mpu6050. We will need to check the datasheet.

We can apply the orientation and calibration of imu and mag in the linorobot2_hardware.

https://github.com/hippo5329/linorobot2_hardware/blob/master/firmware/src/firmware.cpp#L308

Like,
X = X_RAW
Y = -Y_RAW
Z = -Z_RAW

closed 2025-02-26 by pitosalas