hippo5329/linorobot2_hardware #58: Gendrv PWM_bits=10

pitosalas, 2025-02-16:
Just for my own info, how did you derive that? I’ve seen 8 bits in the past, not 10. 

Just to be clear I’m trying to learn, not questioning you!

hippo5329, 2025-02-16:
Motor PWM does not need to be very precise. 10 bits or 8 bits does not make much difference.

10 bits comes from upstream config/lino_base_config.h. 8 bits comes from upstream config/custom/dev_config.h.
Either is fine.

Our target PWM freq is 20KHz. With 10 bits, 20000Hz * 2 ^ 10 = 40.96MHz which is fine for esp32/pico. They are capable of 16 bits PWM. But with 16 bits clock requirement is too high.

pitosalas, 2025-02-17:
The number of bits determine the "resolution". In our case, 0 to 1023. But the % of pwm that is on or off is simply the percent if the 0-1023 range,i.e. value / 1023

Is that correct?

hippo5329, 2025-02-18:
Yes. It is correct.

closed 2025-02-26 by pitosalas