hippo5329/linorobot2_hardware #60: IMU AND MICROROS ON PICO2

saberchihawi, 2025-03-06:
after the success of sensors_test when i upload the firmware and the imu is uncomment i can't get microros connection and led pico is still Blinking !

hippo5329, 2025-03-06:
You will need to uncomment the IMU to test IMU in sensor_tests and micro-ros firmware. There is i2c test in the beggining and you can verify if the IMu is connected to the i2c bus.

saberchihawi, 2025-03-07:
Thank you for the guidance on uncommenting the IMU to test in sensor_tests
and micro-ros firmware. I have successfully checked the IMU data with
monitor , but I encountered an issue when uploading the firmware to test
the topic /imu with ROS. I am unable to establish the micro-ROS connection.
Could you kindly provide further assistance on how to resolve this issue?
Thank you for your support.

saberchihawi, 2025-03-07:
Connect micro-ROS agent again and check /imu/data topic the issues on this step i cant get a micro ros connection when the imu is Uncomment

hippo5329, 2025-03-07:
Please post test_sensors output.

saberchihawi, 2025-03-11:
when i upload the firmware after checking imu alone it sucess but the next steps still can't connect to microros and check the imu data

saberchihawi, 2025-03-11:
[image]

saberchihawi, 2025-03-11:
in this step i cant establich a microros connection when the imu is uncomment in pico2 .config

saberchihawi, 2025-03-11:
[image]

saberchihawi, 2025-03-11:
but how can i get the i2c message with pico2 ?
im using 9250 imu it necessary to add the pull up resistor ??
 please help

SergeyNikolay, 2025-09-07:
Problem when IMU is uncommented, connection with micro-ROS is not established. Resistor 2 kOm sda, scl are installed!

[image]

[image]

[image]

[image]
The last picture does not connect micro-ros.