hippo5329/linorobot2_hardware #61: ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0

pitosalas, 2025-03-06:
I just went back to re-look at previous steps. I have a successful firmware upload to the ESP32 on GenDrv. I can run linorobot2_bringup successfully. 

But when I go back to an earlier step I get only these two lines of output from micro_ros_agent:

```
[ros2run]: Process exited with failure 1
[dome:~/uros_ws] ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0
[1741280567.271572] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 10
[1741280567.272657] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```

hippo5329, 2025-03-06:
You will need to use ttyUSB0 for esp32.

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-serial-transport-on-devttyusb0

hippo5329, 2025-03-07:
You will ned to specify the baudrate 921600.

closed 2025-03-12 by pitosalas