hippo5329/linorobot2_hardware #63: Need help understanding why robot appears to rotate without mag sensor enabled

pitosalas, 2025-03-12:
```
Enable Magnetometer

If you enable magnetometer device, you will need to enable Magdwick filter by adding "madgwick:=true" in robot bringup. The topic from imu will be named as imu/data_raw. The Madgwick filter will fuse imu/data_raw and imu/mag to imu/data. The hippo5329/linorobot2 has been updated with Madgwick filter support in the imu_tools package. You may visualize the IMU and manetometer by [enable the IMU and magetometer plug-ins](https://automaticaddison.com/how-to-publish-imu-data-using-ros-and-the-bno055-imu-sensor/) in RVIZ2.

The EKF (ekf.yaml) takes vx vy yaw of /odom/unfiltered and yaw of /imu/data to compute the /odom. The ekf filter configuration will need update to enable magnetometer, as only 'vyaw' is enabled in the default configuration. Both IMU and magnetometer must be calibrated, otherwise the robot's pose will rotate.
```
Recall I am using gendrv. As you predict, before I enable the magnetometer, in rviz, the robot rotates. Your default ekf.yaml sets ekf filter to get yaw from  both odom and imu. As the robot is stationary, I would expect both to be zero. Certainly the one from odom. So I don't understand why I see a rotation . Thanks/

hippo5329, 2025-03-13:
If you want to disable mag in gendrv, you will need to comment out the line "#define USE_AK09918_MAG" in config file.

https://github.com/hippo5329/linorobot2_hardware/blob/master/config/custom/gendrv_config.h#L40

The default ekf.yaml is fine. Do not change it.

To bring up, use

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-serial-transport-and-lidar-on-the-robot-computer

Do not add madgwick filter.

Check the /imu/data topic and make sure 

angular_velocity x,y,z are all very close to zero.

linear_acceleration x,y are very close to zero.  z is very close to 9.8.

pitosalas, 2025-03-13:
Thanks. I will try this specifically and see how it goes.

Did you see my long post on Home Brew Robotics? If you can correct any wrong points there I think it would be interesting.

hippo5329, 2025-03-14:
Leveling of the gendrv/sensors board is critical. If it tills, the accelerator reading will be incorrect.

pitosalas, 2025-03-14:
It is level. But it is upside down. However I corrected that with the imu calibration. Which by the way, do you mention IMU calibration in your writeup?

Also I would love to understand how you arrive at these settings for ekf. Again, I am not saying that it is wrong just would love to learn :)

        odom0: odom/unfiltered
        odom0_config: [false, false, false,
                       false, false, false,
                       true, true, false,
                       false, false, true,
                       false, false, false]

        imu0: imu/data
        imu0_config: [false, false, false,
                      false, false, false,
                      false, false, false,
                      false, false, true,
                      false, false, false]

You use only vx and vy From odom. Not yaw or vyaw.

And you use only vyaw from imu. Nothing else.

How did you figure that out?

hippo5329, 2025-03-15:
I didn't do imu calibration with ros2 package. The linorobot2_hardware firmware will do angular velocity calibration after reset. The i2cdevlib mpu6050 driver will do both angular velocity and linear acceleration calibration.

https://github.com/hippo5329/linorobot2_hardware/blob/master/firmware/lib/imu/imu_interface.h#L79
https://github.com/hippo5329/linorobot2_hardware/blob/master/firmware/lib/imu/default_imu.h#L126

The default ekf.yaml comes from upstream linorobot2. For odom, vyaw can be derived from vx and vy. For imu, linear acceleration error will be integrated and may result in rotation. So they are dropped. The is a more primitive and conservative configuration. I see some other robot projects use different configurations. I have tried to enable more entries and tune the covariance. I don't remember if I got better results.

I have invited you to be collaborator of my forks of linorobot2 and linorobot2_hardware projects. I would be nice if you could document your usage of imu calibration and improve the wiki guide. Thank you.

pitosalas, 2025-03-15:
Got it. I am going on a short vacation so I won't be able to get to documenting the calibration but I will do it when I return.