hippo5329/linorobot2_hardware #66: i2c on ESP32 interfering with i2c on Pi?

pitosalas, 2025-04-01:
I am chasing a weird problem. I am using the Gendrv board, and using it with a Raspberry Pi, as normal. 

Now, I have connected an i2c breakout board to the correct pins on the Pi, and furthermore connected a simple OLED display through the i2c. This works correctly if the Pi is not mounted on the Gendrv. It also works correctly if the Pi is mounted on Gendrv.

But when I run linorobot2_bringup/bringup.launch.py the  i2c interface gives an error.  

Any insight on what is happening and how I can fix it?

hippo5329, 2025-04-02:
The i2c wires on gendrv are connected to the headers to mount pi. So they can't use at the same time. I would believe this is a design flaw.

You may use the other pins on pi for l2c led.

pitosalas, 2025-04-02:
Thanks! I thought it might be something like that but two things puzzle me:

1) Why does the problem only appear once I run bringup (on the Pi!
2) If i2c allows multiple devices daisy chained together, why does it not like to see a pi and a esp32 at the same time?

hippo5329, 2025-04-02:
Pi and esp32 are i2c bus masters. It might be i2c bus contention from two bus masters. I2c bus does not support multiple masters.

pitosalas, 2025-04-02:
Any theory on: 1) Why does the problem only appear once I run bringup (on the Pi!

Also, what do you suggest as the better workaround: don't plug the pi into the gendrv, and power it separately OR use "software i2c" - bit banging (?) to reassign the two i2c pins (which I read my affect performance, but no idea if that's true.)

hippo5329, 2025-04-03:
The i2c bus contention may break the imu driver on esp32, then the micro-ros failed to connect.

You may unplug the pi and power the pi separately. Or cut the i2c wire near the 40 pins header on the gendrv.

pitosalas, 2025-08-27:
I solved this problem but forgot to explain it here. My raspi is attached to the Gendrv directly via the 40 pin connector. I then try to access i2c via that connector. But the way the Gendrv is designed it is acting as an I2C "master" (right term?) for the IMU. So when I try to attach another i2c board to that same 40pin connector they interfere with each other. Solution is: Dont physically attach (plug in) your Raspi into the Gendrv's 40 pin connector if you want to have additional i2c. devices on the Pi

closed 2025-08-27 by pitosalas