hippo5329/linorobot2_hardware #77: Are these topic hz measurements significant?

pitosalas, 2025-08-27:
I am running a GenDrv and a Raspi 4. Here are the typical hz that I get (with all defaults afaik):

^C[dome:~/linorobot2_ws] ros2 topic hz  /scan
average rate: 10.001
	min: 0.100s max: 0.100s std dev: 0.00020s window: 12

[dome:~/linorobot2_ws] ros2 topic hz  /imu/data_raw
average rate: 37.786
	min: 0.026s max: 0.030s std dev: 0.00117s window: 39

^C[dome:~/linorobot2_ws] ros2 topic hz  /odom
average rate: 14.990
	min: 0.066s max: 0.068s std dev: 0.00057s window: 17

All significantly below what you say I should see. Below output from Top shows cpu is not overloaded:

```
top - 18:54:45 up 18 min,  3 users,  load average: 1.52, 1.17, 1.45
Tasks: 180 total,   1 running, 179 sleeping,   0 stopped,   0 zombie
%Cpu(s):  4.8 us,  5.9 sy,  0.0 ni, 88.9 id,  0.2 wa,  0.0 hi,  0.1 si,  0.0 st
MiB Mem :   7802.4 total,   2798.1 free,   2734.7 used,   2584.6 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   5067.7 avail Mem

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
   2339 pitosal+  20   0 1413328  26264  17816 S  16.2   0.3   2:28.73 micro_ros_agent
   2342 pitosal+  20   0  826660  29436  20988 S   6.9   0.4   1:02.85 ldlidar_stl_ros
   2338 pitosal+  20   0  838656  36808  26184 S   5.3   0.5   0:48.57 ekf_node
   2340 pitosal+  20   0  883032  71756  33320 S   4.3   0.9   0:37.76 joint_state_pub
     11 root      20   0       0      0      0 I   1.7   0.0   0:07.30 kworker/u8:0-events_unbound
   1736 pitosal+  20   0   52.1g 760240  56704 S   1.3   9.5   1:36.04 node
    513 root       0 -20       0      0      0 I   1.0   0.0   0:08.44 kworker/u9:1-brcmf_wq/mmc1:0001:1
   2341 pitosal+  20   0  757528  32008  23176 S   1.0   0.4   0:12.78 robot_state_pub
   2978 pitosal+  20   0   12156   5504   3328 R   1.0   0.1   0:00.10 top
     37 root      20   0       0      0      0 I   0.7   0.0   0:08.33 kworker/u8:1-events_unbound
   2113 pitosal+  20   0   16392   6420   4736 S   0.7   0.1   0:05.30 sshd
   2343 pitosal+  20   0  758480  32304  23728 S   0.7   0.4   0:08.82 scan_to_scan_fi
   2444 pitosal+  20   0  879808  69520  32352 S   0.7   0.9   0:06.14 python3
   2807 root      20   0       0      0      0 I   0.7   0.0   0:00.20 kworker/0:2-events
     84 root      20   0       0      0      0 I   0.3   0.0   0:01.34 kworker/1:2-events
   1623 pitosal+  20   0   16708   6708   4608 S   0.3   0.1   0:02.50 sshd
   1761 pitosal+  20   0  887408  79012  45312 S   0.3   1.0   0:04.79 node
   2201 pitosal+  20   0   14400   6072   3072 S   0.3   0.1   0:02.40 tmux: server
   2564 root      20   0       0      0      0 I   0.3   0.0   0:01.57 kworker/2:2-events
   2794 root      20   0       0      0      0 I   0.3   0.0   0:01.01 kworker/3:0-events
   2795 root      20   0       0      0      0 I   0.3   0.0   0:01.85 kworker/0:0-events
   2798 root      20   0       0      0      0 I   0.3   0.0   0:02.82 kworker/u8:2-events_unbound
      1 root      20   0   22628  12612   8516 S   0.0   0.2   0:03.41 systemd
      2 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kthreadd
      3 root      20   0       0      0      0 S   0.0   0.0   0:00.00 pool_workqueue_release
      4 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/R-rcu_g
      5 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/R-rcu_p
      6 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/R-slub_
      7 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/R-netns
     12 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/R-mm_pe
     13 root      20   0       0      0      0 I   0.0   0.0   0:00.00 rcu_tasks_kthread
     14 root      20   0       0      0      0 I   0.0   0.0   0:00.00 rcu_tasks_rude_kthread
     15 root      20   0       0      0      0 I   0.0   0.0   0:00.00 rcu_tasks_trace_kthread
     16 root      20   0       0      0      0 S   0.0   0.0   0:00.23 ksoftirqd/0
     17 root      20   0       0      0      0 I   0.0   0.0   0:02.57 rcu_preempt
     18 root      rt   0       0      0      0 S   0.0   0.0   0:00.01 migration/0
     19 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/0
     20 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/1
     21 root      rt   0       0      0      0 S   0.0   0.0   0:00.00 migration/1
     22 root      20   0       0      0      0 S   0.0   0.0   0:00.06 ksoftirqd/1
     25 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/2
     26 root      rt   0       0      0      0 S   0.0   0.0   0:00.00 migration/2
     27 root      20   0       0      0      0 S   0.0   0.0   0:00.06 ksoftirqd/2
     30 root      20   0       0      0      0 S   0.0   0.0   0:00.00 cpuhp/3
     31 root      rt   0       0      0      0 S   0.0   0.0   0:00.00 migration/3
     32 root      20   0       0      0      0 S   0.0   0.0   0:00.07 ksoftirqd/3
     35 root      20   0       0      0      0 S   0.0   0.0   0:00.04 kdevtmpfs
     36 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 kworker/R-inet_
     38 root      20   0       0      0      0 S   0.0   0.0   0:00.00 kauditd
     40 root      20   0       0      0      0 S   0.0   0.0   0:00.00 khungtaskd
```
[dome:~/linorobot2_ws]

hippo5329, 2025-08-28:
The hz readings are slower than expected. It might still work though.

You may disable the i2c magnetometer and current sensor. Comment out these lines.

https://github.com/hippo5329/linorobot2_hardware/blob/master/config/custom/gendrv_config.h#L40
https://github.com/hippo5329/linorobot2_hardware/blob/master/config/custom/gendrv_config.h#L221

Maybe I should have them disabled by default.

pitosalas, 2025-08-28:
Hi Thomas,

I will try commenting those two out and rebuilding.

From my side, what’s the fix do you think? Raspi 5? Or something else?

Thanks,

Pito

> The hz readings are slower than expected. It might still work though.
> You may disable the i2c magnetometer and current sensor. Comment out these lines. [...]

Best,

Pito

Boston Robot Hackers && 
Comp. Sci Faculty, Brandeis University

hippo5329, 2025-08-28:
Remove these i2c sensors will reduce the i2c bus loading.

Raspi 5 is much faster then Raspi 4. Be sure to have a fast write speed micro-SD, 50MB/s or better. I have tried USB SSD on the USB 3.0 port of Pi 5. It is mcuh faster than micro-SD.

pitosalas, 2025-08-28:
I also have an i2c “shim” directly on the Raspi for an lcd display. Could that be a problem? The code supporting the display is NOT running in this scenario.

> 
> Remove these i2c sensors will reduce the i2c bus loading.
> Raspi 5 is much faster then Raspi 4. Be sure to have a fast write speed micro-SD, 50MB/s or better. I have tried USB SSD on the USB 3.0 port of Pi 5. It is mcuh faster than micro-SD.

Best,

Pito

Boston Robot Hackers && 
Comp. Sci Faculty, Brandeis University

pitosalas, 2025-08-28:
Making the two i2c changes helped a little. But those numbers are so "round" (9.995 and 15.002) it makes me think it's something else. I assume the expected rate should be set by:

// #define CONTROL_TIMER 20

Which is not set in the include files and set to 20 in firmware.cpp:

#ifndef CONTROL_TIMER
#define CONTROL_TIMER 20 // 50Hz
#endif

Weird. any other ideas?

[dome:~/linorobot2_ws] ros2 topic hz /imu/data_raw
average rate: 43.324
	min: 0.022s max: 0.026s std dev: 0.00104s window: 45

[dome:~/linorobot2_ws] ros2 topic hz /scan
average rate: 9.995
	min: 0.100s max: 0.101s std dev: 0.00037s window: 12

[dome:~/linorobot2_ws] ros2 topic hz /odom
average rate: 15.002
	min: 0.065s max: 0.068s std dev: 0.00058s window: 16

hippo5329, 2025-08-28:
/scan is driven by lidar ld19. ~10Hz

/odom is output of ekf filter, the freq is specified in ekf.yaml, which should be ~50Hz.

You may connect gendrv to pc

pitosalas, 2025-08-28:
All set:

```
================================================================================
ROS2 Topic Hz Monitor - 16:44:02
================================================================================
Topic                                    Type                      Hz         Count
--------------------------------------------------------------------------------
/scan                                    LaserScan                  11.66     24
/imu/data_raw                            sensor_msgs/msg/Imu        44.94     90
/odom                                    nav_msgs/msg/Odometry      41.82     84
/tf                                      tf2_msgs/msg/TFMessage     51.62     102
================================================================================
```

closed 2025-08-28 by pitosalas